wradlib/wradlib #17: Python3 - Warning when importing wradlib

Thomas1980nl, 2016-03-03:
Hello,

After installing wradlib and all dependencies in python3, I get the following error when importing wradlib:

>>> import wradlib

Warning (from warnings module):
  File "/usr/lib/python3/dist-packages/numpy/oldnumeric/__init__.py", line 11
    warnings.warn(_msg, ModuleDeprecationWarning)
ModuleDeprecationWarning: The oldnumeric module will be dropped in Numpy 1.9

I am wondering what it means and what I can do to solve this. Thanks in advance for your help!

Best regards,
Thomas

PS: easy_install3 crashed when trying to install wradlib, so I did a manual installation.

kmuehlbauer, 2016-03-04:
@Thomas1980nl 

wradlib has enabled deprecation warnings by default (see [wradlib/.__init__.py](https://github.com/wradlib/wradlib/blob/master/wradlib/__init__.py)). This is a DeprecationWarning from the `numpy` package. This shouldn't have impact on wradlib's functionality. 

May I ask, on which OS and which python you are working and which installation process you ran (pip, python setup.py etc. how you did install the dependencies? We have major restructuring going on, and I want to be sure of the version we are talking about.

Thomas1980nl, 2016-03-04:
Thanks for your quick reply. Good to know that this should not have any impact on wradlib's functionality.

For your info:
I am using Linux (Ubuntu 14.04, 32 bit) and python 3.4.3. I tried to install wradlib using easy_install3 after installing the dependencies that were not yet installed manually (most of them via apt-get install). But easy_install3 crashed when trying to install wradlib. Then I downloaded the latest version (0.7.2) manually, and installed it using python3 setup.py.

Thanks for your work on the library!

Best regards,
Thomas

kmuehlbauer, 2016-03-04:
The version on Pypi (0.7.0) isn't Python3 compatible. So this is obviously the reason for the crash. 

We are currently in the last stages of reaching python3 compatibility. It's planned to release version 0.8.0 next week. Stay tuned.

Thomas1980nl, 2016-03-04:
Problem solved! I will keep an eye out for version 0.8.0.

Thanks for your help.

kmuehlbauer, 2016-03-04:
If you want, you can try the preliminary releases on `testpypi` (0.7.5 I think)

kmuehlbauer, 2016-03-10:
@Thomas1980nl v0.8.0 is available on [pypi](https://pypi.python.org/pypi/wradlib/0.8.0)

Closing

closed 2016-03-10 by kmuehlbauer