wradlib/wradlib #25: Bug in ZonalDataBase when reading in layers from file using load_vector method

heistermann, 2016-03-08:
I think there is a bug in `ZonalDataBase.loadvector` (ll. 546 ff). Instead of

`ogr_copy_layer(ds_in, 0, self.src.ds)`
`ogr_copy_layer(ds_in, 1, self.trg.ds)`
`ogr_copy_layer(ds_in, 2, self.dst.ds)`

it should be

`ogr_copy_layer(ds_in, 0, self.dst.ds)`
`ogr_copy_layer(ds_in, 1, self.src.ds)`
`ogr_copy_layer(ds_in, 2, self.trg.ds)`

At least on my system... otherwise, you will have an error once you try to read `ZonalData` that has been dumped to a file before because the layers get mixed up. Can anyone reproduce this?

kmuehlbauer, 2016-03-08:
Yes, this is a bug. The above is OK, but the error is here:

```
        # create all DataSources
        self.src = DataSource(name='src')
        self.src.ds = drv_in.CreateDataSource('src')
        self.trg = DataSource(name='trg')
        self.trg.ds = drv_in.CreateDataSource('trg')
--->  self.dst = DataSource(name='trg') # needs to be 'dst'
        self.dst.ds = drv_in.CreateDataSource('dst')

        # copy all layers
        ogr_copy_layer(ds_in, 0, self.src.ds)
        ogr_copy_layer(ds_in, 1, self.trg.ds)
        ogr_copy_layer(ds_in, 2, self.dst.ds)

        # get spatial reference object
        self._srs = self.src.ds.GetLayer().GetSpatialRef()
        self.src._srs = self.src.ds.GetLayer().GetSpatialRef()
        self.trg._srs = self.trg.ds.GetLayer().GetSpatialRef()
--->  self.dst._srs = self.trg.ds.GetLayer().GetSpatialRef() # needs to be self.dst.ds.Get...
```
@heistermann  Could you check this?

kmuehlbauer, 2016-03-08:
@heistermann Trying to reproduce, do you have a minimaö example showing the error on your system. Maybe there is more than what I mentioned above.

heistermann, 2016-03-09:
Don't have a minimal example, yet. But the way to reproduce the error is to use `examples/tutorial_zonal_statistics.py`. Run it once and it will work fine because intersections are produced from scratch. Run it again and it'll fail trying to read the data from the shapefiles:

`RuntimeError: 'trg_index' not recognised as an available field.`

Fixing the bug that you pointed out above does not fix the problem. My approach to understand the problem was to move to after line 532 in debug mode:

`ds_in, tmp = io.open_shape(filename)`

Then I tried to understand how the layer indices work. Here is the output:
`>>> ds_in.GetLayerByIndex(0).GetName()`
`'dst'`
`>>> ds_in.GetLayerByIndex(1).GetName()`
`'src'`
`>>> ds_in.GetLayerByIndex(2).GetName()`
`'trg'`

My suspicion is that layer indices are not assigned in the order the layers were created (in ``dump_vector`), but e.g. alphabetically. So in order to fix the problem, We also need to do:

`ogr_copy_layer(ds_in, 1, self.src.ds)`
`ogr_copy_layer(ds_in, 2, self.trg.ds)`
`ogr_copy_layer(ds_in, 0, self.dst.ds)`

I would also suggest to modify the example so it can routinely test how intersections are loaded. Right now, it is only run once and thus will not capture this failure. I'll would prepare a PR that basically removes the try/except and forces the example to first dump and than again load.

kmuehlbauer, 2016-03-09:
@heistermann If I run the example twice, I get no error. Maybe the gdal driver has some other output on different OS. Are you on windows?

The `ogr_copy_layer` funcion gets the layer name from the index. So we easily could read the layers by name in the first place. There had to be a reason, why I did it like this, but this may now not be necessary any more. So I suggest to have some `ogr_copy_layer_by_name` function where an index isn't necessary.

heistermann, 2016-03-09:
Interesting.

But I totally agree with your suggestion to use GetLayerByName. I'll prepare a PR.

kmuehlbauer, 2016-03-09:
Yes, very interesting. Any need to investigate this further? We should keep that in mind, just in case anything unusual is happening again.

Do you change my other mistakes when preparing PR?

kmuehlbauer, 2016-03-09:
Could you also replace the `exit(0)` in the `has_geos()`-check with just an `return` in both the zonalstats examples?

I found that `exit(0)` raises SystemExit and breaks the unittest, with just return, the test passes with the respective warning written to screen.

kmuehlbauer, 2016-03-09:
Sorry, I made the exit(0)/return changes in #22 to not break the unittests in travis.

PR #28 (2016-03-09): Fixing some issues in zonalstats that appear to be Windows specific.
Without understanding what actually is behind this behaviour, this PR shold fix issue #25.

It also makes sure that the examples test whether loading ZonalData from file actually works. 

closes #25
closed 2016-03-09 by kmuehlbauer