wradlib/wradlib #26: gdal - geos issues with conda

kmuehlbauer, 2016-03-09:
At the moment the travis tests do not run properly for python2.7 and python3.4 because the conda provided gdal has geos not compiled in. The python35 works fine.

How do we tackle this?

We could compile our own conda packages with all needed dependencies, but this may not be short term.

heistermann, 2016-03-09:
I don't know how the conda environment works. But would it be possible to combine our "old" way of pip-installing gdal with the conda approach? Or is the geos component not included this way, either?

kmuehlbauer, 2016-03-09:
The problem is not with the python bindings of gdal but with `libgdal`.  Anyway we need also geos (or libgeos in our case) because libgdal is dynamically loading libgeos-functions. 

I did some tests here to compile gdal/gdal-python (incl. geos) locally for conda. This really works, but I do not have the complete picture, yet.

jjhelmus, 2016-03-09:
There may be a version of gdal/libgdal on [Anaconda.org](https://anaconda.org/search?q=type%3Aconda+gdal) which has proper support for Python 2.7 and 3.4.  I recall that the packages in the OSGEO repository were decent.  I'll investigate this locally on my machine.

A more maintainable solution would be to have a GDAL package built using the conda-forge setup.  It looks like you started this process already in conda-forge/staged-recipes#82.

kmuehlbauer, 2016-03-09:
@jjhelmus Yes, from my perspective the conda-forge approach is best.

jjhelmus, 2016-03-09:
Turns off the the gdal packages provided by Continuum do support GEOS but the libgdal 2.0.0_0 package is broken and installed by default in Python 2.7 and 3.4 when gdal is installed.  libgdal 2.0.0_2 works fine.  PR #29 address this by explicitly installing libgdal which insures the latest version and build is installed.

kmuehlbauer, 2016-03-09:
@jjhelmus Thanks for looking into this. Seems that this fixes it for now. Anyway we should go the code-forge way, if that is available.

PR #29 (2016-03-09): Fix GDAL issue on Travis CI
Explicitly conda install libgdal on Travis CI to insure the latest version of libgdal is install in the conda environment.

closes #26
closed 2016-03-09 by kmuehlbauer