wradlib/wradlib #27: Release v0.8.0

kmuehlbauer, 2016-03-09:
Most issues for v0.8.0 are fixed. 

Work is in PRs (#5, #9, #10-15, #18-22).

At the moment we have only #25 which need to be adressed before release of v0.8.0.  So if agreed, we could release v0.8.0 after fixing #25. Any comments?

heistermann, 2016-03-10:
I agree - it'll be good to have all these major changes wrapped into a fresh release, even if some issue might still to be found. Maybe we'll have some ore bug fix releases in the 0.8.x series.

I have to say you did some outstanding work to make all these transitions happen.

kmuehlbauer, 2016-03-10:
Thanks @heistermann, it was quite a lot of work, but in the end we all will benefit from the workflow. 

As release 0.8.0 will hit the community as the first wradlib python3 compatible release this surely will rebound in kind of bug reports, questions, enhancement requests and hopefully code additions.

I have also started the [wiki here](https://github.com/wradlib/wradlib/wiki) with instructions for creating a wradlib release.

First point there is to update the doc `release_notes.rst` with the latest changes. I can prepare the necessary PR. In this PR also the switch from tespypi to pypi is due. 

For the mailing list and the wradlib.org web page a bit more comprehensive writeup is needed imho. @heistermann could you take the lead with this? My suggestion is to create the article on the wradlib.org homepage and refer to it on the mailing list. This way we can also announce the new domain.

If there is more for the todo (see above) just add it there.

heistermann, 2016-03-10:
Great wiki article - very helpful. Until now we did not have such a transparent documentation of the realease workflow.

I can draft an article to accompany the release of 0.8.0. Watch out for a PR at wradlib/landing-page.

kmuehlbauer, 2016-03-10:
The prerelease PR  #30 is ready for merge.

@heistermann Ready to proceed to release version v0.8.0, if you agree.

heistermann, 2016-03-10:
Go ahead.

For anouncement of the new release, I drafted an article - please review wradlib/landing-page#4.

wradlib, 2016-03-10:
Release tag created, travis running, fingers crossed.

wradlib, 2016-03-10:
Well, it's done:

* all tests are fine
* we have [v0.8.0 on pypi](https://pypi.python.org/pypi/wradlib/0.8.0)
* we have the docs out http://wradlib.org/wradlib-docs/0.8.0/

Thanks to all collaborators, @heistermann, @jjhelmus, @kmuehlbauer 

Finally, **"All's well that ends well"**

heistermann, 2016-03-10:
Congrats, @kmuehlbauer !

kmuehlbauer, 2016-03-10:
Everything went really smooth now, closing.

closed 2016-03-10 by kmuehlbauer