wradlib/wradlib #58: Citations

kmuehlbauer, 2016-04-22:
With the docs our [bibliography](http://wradlib.org/wradlib-docs/latest/zreferences.html) is created.

For the handling of citations within notebooks I propose to switch from the numbering to alphanumeric citing. This means to change

```
[12]	S. Jacobi, M. Heistermann, and T. Pfaff. Evaluation and improvement of C-band radar attenuation
correction for operational flash flood forecasting. In Proceedings of weather radar and hydrology 
symposium in Exeter, UK, number 351, 33–38. IAHS, 2012.
```
which got the [12] as hyperlink to `[JH15]`.

Why do this? 

With this we can do a nice markdown link like so [JH15](http://wradlib.org/wradlib-docs/latest/zreferences.html#jacobi2015) within the notebook. This will in any case be rendered also in the final doc and point to the right place: http://wradlib.org/wradlib-docs/latest/zreferences.html#jacobi2015. 

With the normal numbering, this can change if we drop in some paper.

kmuehlbauer, 2016-07-19:
ping @heistermann How is you opinion on this? 

With the alphanumeric citing, we can avoid re-ordering (re-numbering) and therefore can retain linkage in the future.

heistermann, 2016-07-20:
I agree. What would you think about using standard in-text citation format for hyperlinking - such as  [Jacobi et al. 2015][JH15](http://wradlib.org/wradlib-docs/latest/zreferences.html#jacobi2015)? This way, readers could more intuitively anticipate the actual source. However, I understand that the actual bibliography would loose readability because the "numbering" would be more bulky.

kmuehlbauer, 2016-07-20:
There are two different things, we have to consider, the .processing of `.rst` files and processing of `notebooks`

- We have the `refs.bib` file, where all our references are listed in bibtex-format. Then we have the zreferences.rst, where this bibtex-file is incorporated into docs. While *sphinxing* the `.rst`-files (there are not that much any more, but we have quite a bunch in the module docstrings), all `:cite:`-directives are translated to some hyperlink to the appropriate reference in the final doc.

- This doesn't work for notebooks, because the transformation/sphinx-process is different. If we want to refer to our references-html within the standalone-notebook, we need something like [Jacobi et al. 2015](http://wradlib.org/wradlib-docs/latest/zreferences.html#jacobi2015), which points to `latest` in any case. And, this will also work for the compiled doc, but also older releases of the doc, will point to latest.

I fully agree with your proposal to have a more clear citation format. We just have to find out how to tweak `sphinx` to use `Jacobi et. al` is displayed reference item.

kmuehlbauer, 2016-07-20:
I'm not sure, if we get this to work with the current `sphinxcontrib-bibtex` extension. 

We might consider [sphinx-natbib](http://wnielson.bitbucket.org/projects/sphinx-natbib/) which has a greater flexibility.

kmuehlbauer, 2016-07-20:
> We might consider sphinx-natbib which has a greater flexibility.

Just saw that this project seems abandoned (latest changes from 2012) :frowning: .

kmuehlbauer, 2016-07-20:
@heistermann 

OK, I think I have a solution, but want to be sure, we are referring to the same look and feel.

For now, we have `.rst`-files with:

Goudenhoofdt [[8]](http://localhost:63342/wradlib/docs/zreferences.html#goudenhoofdt2009) provide an excellent overview of adjustment procedures. 

This would look like this with the new style in the compiled .rst (the leading `Goudenhooft` removed). 

[[Goudenhoofdt et al., 2009]](http://localhost:63342/wradlib/docs/zreferences.html#goudenhoofdt2009) provide an excellent overview of adjustment procedures. 

Anyway it would also look like this, on the zreferences.html

|      |      |
| --- | --- |
| [Goudenhoofdt et al., 2009] | E. Goudenhoofdt and L. Delobbe. Evaluation of radar-gauge merging methods for quantitative precipitation estimates. Hydrology and Earth System Sciences, 13(2):195–203, 2009. doi:10.5194/hess-13-195-2009. |

Shall we go for it?

For the notebooks, we can do everything we want, like:

- [[Goudenhoofdt et al., 2009]](http://localhost:63342/wradlib/docs/zreferences.html#goudenhoofdt2009) provide an excellent overview of adjustment procedures. 
- [(Goudenhoofdt et al., 2009)](http://localhost:63342/wradlib/docs/zreferences.html#goudenhoofdt2009) provide an excellent overview of adjustment procedures. 
- [Goudenhoofdt et al., 2009](http://localhost:63342/wradlib/docs/zreferences.html#goudenhoofdt2009) provide an excellent overview of adjustment procedures.

heistermann, 2016-07-20:
Perfect!

kmuehlbauer, 2016-07-22:
Done in #74, Closing

closed 2016-07-22 by kmuehlbauer