wradlib/wradlib #59: input array/floats

nguy, 2016-05-06:
Given that wradlib is based upon numpy arrays, would it be worth ensuring that input comes in as a numpy array? For example, in ```wradlib.zr.z2r``` if 
```
>>> z1 = 10.
>>> wradlib.zr.z2r(z1)
0.1537645610180688

>>> z2 = np.array([10., 20., 30.])
>>> wradlib.zr.z2r(z2)
array([ 0.15376456,  0.23713737,  0.30553223])
```
But if a user say put a list in:
```
>>> z3 = [10., 20., 30.]
>>> wradlib.zr.z2r(z3)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/guy/anaconda/lib/python2.7/site-packages/wradlib-0.7.0-py2.7.egg/wradlib/zr.py", line 57, in z2r
    return (z/a)**(1./b)
TypeError: unsupported operand type(s) for /: 'list' and 'float'
```

By using np.asarray, I think you can avoid this:
```
>>> wradlib.zr.z2r(np.asarray(z1))
0.15376456101806879
>>> wradlib.zr.z2r(np.asarray(z3))
array([ 0.15376456,  0.23713737,  0.30553223])
```

kmuehlbauer, 2016-05-09:
@heistermann How should we proceed here?

The docstring says `z` parameter should be float or array of floats. As @nguy wrote, it breaks when giving `z` as a list (as expected). I'm not sure if we really want to allow lists as input parameters.

@nguy Could you explain in a bit more detail why you like to give a list? Do you have an example, where this might be needed?

heistermann, 2016-05-09:
I agree - I do not see a clear need to do type conversion internally. What we could do is to raise a more informative exception message...but this one isn't too bad, either, is it?

nguy, 2016-05-09:
This popped into my head recently looking around at other packages and conversations about file inputs in another package (disdrometer stuff, do different file types). Some folks have seen issues when reading in some data and they need to ensure that the input for calculations were converted to numpy arrays.

That said as @kmuehlbauer points out, this isn't needed if the intrinsic restriction is float or array for the function. I guess normally the data that has been read in by wradlib (or pyart or other for that matter) is a numpy array. My intent was just to see if you thought it were worth putting a "check" in here, though it does increase code complexity (perhaps needlessly?).

kmuehlbauer, 2016-05-10:
@heistermann @nguy 

However, we should be careful on how we are using lists and stuff. Since `+` and `*` works for lists as well as for numpy arrays, we should keep that in mind. Also for python2/3 compatibility we have some issues with lists.

For the most use cases duck-typing is best. On the other hand if we need to have a very specific data layout, then we would need to have checks on that layout. This would also be my suggestion.

Anyway, thanks for raising this issue @nguy.

kmuehlbauer, 2016-06-09:
Feel free to reopen, if this needs further discussion.

closed 2016-06-09 by kmuehlbauer

nguy, 2016-06-09:
Will do, I think you addressed it for now and just keeping this in mind in future developments.
Thanks!