wradlib/wradlib #67: sample code for classify_echo_fuzzy

kojinsan, 2016-07-13:
I would be appreciate if you can provide a sample code for the Kdp based dual-pol rain fall estimation, especially on the usage of classify_ech_fuzzy. Thank very much.
Jin-san Ko

heistermann, 2016-07-13:
Thanks for this suggestion! Please watch the status of this issue in order to stay updated on the implementation progress.

irenemeanspeace, 2016-08-11:
Hi Jin-san,

I made a [notebook](https://nbviewer.jupyter.org/github/irenemeanspeace/phirad/blob/master/clutter/Clutter%20detection%20using%20polarimetric%20approach%20%28EDGE%20netcdf%29.ipynb) on clutter detection using a polarimetric approach, and here it is shown how to use the `classify_echo_fuzzy` function. I'll make another one for the KDP based rainfall estimation.

Hope this helps,
irene

irenemeanspeace, 2016-08-11:
Hi Jin-san,

Here's another [notebook](https://nbviewer.jupyter.org/github/irenemeanspeace/phirad/blob/master/rainfall%20estimation/Rainfall%20estimation%20from%20KDP%20%28without%20clutter%20correction%29.ipynb) showing an example on how to do the kdp-based rainfall estimation.

Let me know if you have any questions.

Cheers,
irene

kojinsan, 2016-08-11:
Thank. I will study it.

從我的 iPad 傳送

heistermann, 2016-08-18:
Thanks, Irene! Im closing this issue. However, we should think about adopting these notebooks in wradlib docs under "Tutorials and Examples". I will look into that.

closed 2016-08-18 by heistermann