wradlib/wradlib #70: Create tree structure for example notebooks and documentation?

heistermann, 2016-07-15:
There will be a lot of notebooks in the notebooks directory, and in the corresponding documentation. Should we think about organizing these in some kind of tree structure according to topics? Something like:

- First steps
- Georeferencing
- Quality control
- Visualisation
- ...

This way, users will have a better overview *where* they can or should look for what they need. We might also have a kind of course structure in the back of our minds when desiging a tree.

kmuehlbauer, 2016-07-15:
Seems very reasonable. :+1: 

The `radolan`-(sub)notebooks have already an higher-level document `radolan.ipynb` from which the sub-notebooks can be reached. This also goes nicely in the docs.

A tree structure would would definitely help users (and devs also). I suggest to tackle this when all notebooks are in place.

heistermann, 2016-07-18:
Very good idea - using "high-level" notebooks to define the directory tree for a specific topic. I also agree to address this once all notebooks are translated.

heistermann, 2016-07-21:
Maybe we can discuss a structure right here. 

These are my suggestions for high-level notebooks and the actual example notebooks that we currently have. Please note:
- some notebooks might fit in different topics and could therefore also be *linked* from different topics (e.g. wradlib_gis_export_example.ipynb might fit under RADOLAN and under File I/O),
- for those sections with a checkmark a high level notebook already exists,
- Some section only contain *one* notebook, yet - but additional notebooks might come in the future.
- Of course, additional sections might be created at any time if it makes sense.

- [ ] **Getting started**
- wradlib_introduction.ipynb
- wradlib_workflow.ipynb
- wradlib_get_rainfall.ipynb
- radolan_quickstart.ipynb

- [x] **RADOLAN: radolan.ipynb**
- radolan_format.ipynb
- radolan_grid.ipynb
- radolan_network.ipynb
- radolan_quickstart.ipynb
- radolan_showcase.ipynb
- wradlib_gis_export_example.ipynb
- simple_plot.ipynb

- [ ] **File I/O): file_io.ipynb**
- wradlib_radar_formats.ipynb
- wradlib_gis_export_example.ipynb
- wradlib_generic_netcdf_example.ipynb
- wradlib_load_rainbow_example.ipynb
- wradlib_reading_dx.ipynb

- [ ] **Georeferencing**
- wradlib_georef_example.ipynb
- wradlib_coords_example.ipynb

- [ ] **Visualisation**
- wradlib_plot_curvelinear_grids.ipynb
- wradlib_plot_ppi_example.ipynb
- simple_plot.ipynb
- wradlib_overlay.ipynb

- [ ] **Clutter and echo classification**
- wradlib_clutter_gabella_example.ipynb
- wradlib_clutter_cloud_example.ipynb
- wradlib_histo_cut_example.ipynb

- [ ] **Attenuation**
- wradlib_attenuation.ipynb

- [ ] **Beam blockage**
wradlib_beamblock.ipynb

- [ ] **Multi-sensor merging**
- wradlib_adjust_example.ipynb

- [ ] **Zonal statistics**
- wradlib_zonalstats_classes.ipynb
- wradlib_zonalstats_example.ipynb

- [ ] **Interpolation**
- wradlib_ipol_example.ipynb

- [ ] **Verification**
- wradlib_verify_example.ipynb

- [x] **Example workflows and recipies: recipes.ipynb**
- recipe1.ipynb
- recipe2.ipynb

- [ ] **For developers**
- wradlib_api_change.ipynb

heistermann, 2016-07-21:
And once we fixed the structure, we should discuss whether we should dissolve the doc sections "Tutorials", "Notebooks", and "Recipies" into one single "Tutorials and examples" section...

kmuehlbauer, 2016-07-21:
I'm fine with this. 

We have actually these notebooks in the notebooks.rst:

```
.. toctree::
   :maxdepth: 2
   
   wradlib_introduction.ipynb
   wradlib_api_change.ipynb
   wradlib_beamblock.ipynb
   wradlib_overlay.ipynb
   simple_plot.ipynb
```

Then let's do this now! I'have some changes for the notebooks on handling admonitions. As we do not want to be trapped in a dead end, I fixed some things to be in alignmentt with `nbsphinx` latest changes.

I can do the proposed changes within my upcoming PR. Thoughts?

kmuehlbauer, 2016-07-21:
OK, I'm about to create the following subfolders, below wradlib/notebooks:

attenuation
beamblockage
classify
develop
fileio
georeferencing
interpolation
multisensor
radolan
verification
visualisation
workflow
zonalstats

The high-level notebooks go into the parent folder, all other notebooks into the corresponding subfolder. Do you agree? If we can settle on this, I would also fix the linking.

heistermann, 2016-07-21:
Sorry for the late reply. Great - go ahead!

kmuehlbauer, 2016-07-21:
Will establish the same structure within the doc/source-folder to keep inter notebook references.

kmuehlbauer, 2016-07-21:
Heavy changes here. One thing @heistermann, should we package the rendered notebooks with PyPi? This is just a change of two lines of code.

heistermann, 2016-07-21:
I'd prefer to distribute them unrendered. If people want to look at the rendered notebooks, they can look at the docs. But for courses, I think it is better if the notebooks do not already show the results.

heistermann, 2016-07-22:
How you pulled through with the implementation of this issue - that was really an outstanding effort, Kai! I was just looking at the new docs - I am really happy with the results!

So should we close this issue? - from now on, it's a continuous effort, I guess.

kmuehlbauer, 2016-07-22:
I have the follow-up PR #79 ready, so we can close this here.

closed 2016-07-22 by kmuehlbauer