wradlib/wradlib #72: can you give a example about OrdinaryKriging in ipol ?

shanyuk, 2016-07-16:


heistermann, 2016-07-18:
Thanks for the suggestion - we will provide an example as soon as possible.

kmuehlbauer, 2016-07-23:
@shanyuk You can reference code lines like this (You just need to click on the line number and copy the address from the browser):

https://github.com/wradlib/wradlib/commit/a7e899d1955e4ffeecc504ad0eb1fa1685522e67#diff-2732686ff11606e709ced2a8be7afaa7R501

Then your comment can be inside the corresponding issue and is not without reference.

@heistermann is very busy atm and might not have any cycles to look into this the next couple of days. I can't be of much help, but there might be a chance that @thpfaff can assist.

heistermann, 2016-08-18:
@shanyuk: Does the problem still occur - could you please provide a full example to reproduce the error (including the data if necessary - or ideally with code that generates synthetic data).

Apart from that, you can now find a very simple application example of OK in [wradlib's interpolation tutorial] (http://wradlib.org/wradlib-docs/latest/notebooks/interpolation/wradlib_ipol_example.html).

shanyuk, 2016-09-05:
Think you. I have solved the problem with the example.

kmuehlbauer, 2016-09-05:
Thanks for the feedback, @shanyuk!

closed 2016-09-05 by kmuehlbauer