wradlib/wradlib #83: Finalize Release 0.9.0

kmuehlbauer, 2016-07-22:
Steps to be done to finalize release 0.9.0:

- [ ] Fix release notes (add changelog)
- [ ] Find and fix inter-doc-link issues (docstrings, eg. links to generated-api-functions )
- [ ] review Tutorials and Examples doc-section (headings, toc etc)
- [ ] repair Note, Seealso, Warning in Notebooks (see [wradlib_workflow](https://github.com/wradlib/wradlib/blob/master/notebooks/basics/wradlib_workflow.ipynb) for guidance, but do not use github renderer for evaluation, it's broken in some way)  
- please add other necessary steps here or in the comments below

heistermann, 2016-08-19:
In the TOC of "Tutorials and Examples", some headers lack a whitespace inbetween words (example: http://wradlib.org/wradlib-docs/latest/notebooks.html > "Typical Workflow for Radar-Based RainfallEstimation"). I think these are all from section headers inside notebooks. I have no clue about the cause - the raw headers in the notebooks look fine to me. However, this should not be a showstopper for this release.

kmuehlbauer, 2016-08-21:
> I have no clue about the cause - the raw headers in the notebooks look fine to me. However, this should not be a showstopper for this release.

Me neither. I'll have some checks to hopefully find the cause of this.

heistermann, 2016-08-26:
@kmuehlbauer: Is there anything you would still need for this release? One check mark is still missing at `review Tutorials and Examples doc-section`, but I did not find anything serious, and I think it is an ongoing issue anyway.

kmuehlbauer, 2016-08-26:
@heistermann I'd love to get out 0.9.0 asap. It is all very much done now, especially the addition to examples and tutorials. We only need to get all changes into the changeling (most is already in,  but the latest Bugfixes are missing there). So we can move forward and release 0.9.0.

heistermann, 2016-08-26:
See Pull Request #97 for updating the release notes.

kmuehlbauer, 2016-08-29:
OK, I think we are done here. 

@heistermann If you agree, I'll take the necessary steps to create wradlib release `0.9.0` asap.

heistermann, 2016-08-29:
Absolutely yes. Thanks in advance!

kmuehlbauer, 2016-08-29:
There was some glitch in .travis.yaml pointing devel and release to testpypi. So the release was uploaded to testpypi. I manually uploaded it to pypi-too. A PR which fixes this behaviour follows.

closed 2016-08-29 by kmuehlbauer