wradlib/wradlib #86: Ipol breaking due to changes in scipy 0.18.0

kmuehlbauer, 2016-07-26:
When using scipy 0.18.0 the ipol notebook breaks while handling nan-data. Needs reworking. Pinned scipy to 0.17.1 within travis in #84.

kmuehlbauer, 2016-07-26:
See this [travis-log](https://travis-ci.org/wradlib/wradlib/jobs/147456764).

kmuehlbauer, 2016-08-23:
@heistermann Did you had the chance to look into this? Any ideas?

heistermann, 2016-08-24:
@kmuehlbauer: Not yet, but asap. Thanks for pointing this out.

heistermann, 2016-08-24:
Apparently related to this scipy/scipy#6515

Problem occurs in case the parameter `nnearest` is larger than the number of source points. I'll try to find a more generic solution.

kmuehlbauer, 2016-08-24:
There were some changes in ckdtree implementation, but I‘m not sure, if this is effecting this. Let‘s see what the scipy devs say.

heistermann, 2016-08-25:
Addressed by Pull Request #95.

closed 2016-08-25 by heistermann