wradlib/wradlib #88: MemoryError with wradlib.io.read_RADOLAN_composite

itati01, 2016-07-27:
Dear all,

I am using wradlib 0.8.0 with Python 2.7.10 under Windows 7 Professional (32bit). Unfortunately, wradlib.io.read_RADOLAN_composite() is throwing a MemoryError when reading many binary Radolan RW files.

> File "C:\Python27\lib\site-packages\wradlib\io.py", line 864, in read_RADOLAN_composite
>    arr = arr * attrs["precision"]

The gzipped Radolan files are in separate folders for the years 2008 to 2015. The task is to read all 70000+ files, to apply the Gabella filter function, and to storing the files in one netcdf. The error occurs after a few thousand loops. The exact number of loops varies. Strangely, I successfully managed to store all files without the clutter removal.

```
clutter = wradlib.clutter.filter_gabella(data)
data1 = wradlib.ipol.interpolate_polar(data, clutter) 
```
It is my first report of an issue. If you need more specific information, I am happy to provide it. Currently, I am trying to create annual netcdfs which I will stack afterwards. 

Best regards,
Andreas

kmuehlbauer, 2016-07-27:
Thanks for the report @itati01.

Could you please provide the full error. Which RADOLAN products are you using. We have several years of RADOLAN data available and I could test your code on a linux system. There might be an issue with file handles which get consumed. But that's just a guess.

itati01, 2016-07-27:
Thanks for the quick reply. What do you mean with "full error"? The traceback says

> Traceback (most recent call last):
>   File "..\radolan2nc_subset_.py", line 120, in <module>
>     data, attrs = wradlib.io.read_RADOLAN_composite(f)    # read binary data
>   File "C:\Python27\lib\site-packages\wradlib\io.py", line 864, in read_RADOLAN_composite
>     arr = arr * attrs["precision"]
> MemoryError

The product is Radolan RW, downloaded from the DWD server, untarred and gzipped were necessary (to avoid low disk space). As I said, I happened to create the netcdf once on the same laptop and was surprised to see this error. Btw, the same script is running on our server to test it there. I could report the results on Friday.

Andreas

kmuehlbauer, 2016-07-27:
> Which RADOLAN products are you using.

Should read more carefully. And yes, we have also RW data available (think they are also provided free of charge at dwd ftp).

kmuehlbauer, 2016-07-27:
Can you check the memory consumption while running this? The memory on your 32bit machine might just run out. Yes, it's guess work.

I did similar things, just converting RX data to monthly netcdf. Never got any issues, though. 

BTW, how did you install wradlib? We have some recent source distributions on testpypi.

itati01, 2016-07-27:
Kai,

good guess! It is a memory leak. While *successfully* running the script without clutter removal for 2008, I observed a constant memory consumption of 150 MB for the python task and 1.5 GB in total (all numbers taken from the Windows task manager). *With* clutter removal, the memory consumption of the python task is rapidly growing! After ca 1000 files, python allocates >750 MB (poor server...). After >1GB I terminated the script.

I used pip to install wradlib. If you have any version for testing, I would be glad to test it here.
Andreas

itati01, 2016-07-27:
Kai,
I installed the version '0.8.3.dev+96d3dcb' from testpypi and observed the same problem. I could narrow down the memory leak to ipol.interpolate_polar(). When deactivated, the memory consumption is stable.
Andreas

kmuehlbauer, 2016-07-27:
@itati01 You found the latest test-version :+1: 

> With clutter removal, the memory consumption of the python task is rapidly growing! After ca 1000 files, python allocates >750 MB (poor server...). After >1GB I terminated the script.

OK, then we have to look for the memory hole. It might be within wradlib, or in the way your iteration loop is designed. So sharing the code for testing would be a good idea to track this down. You can send it by email, if you like.

By quickly looking over the implementation of `interpolate_polar`, I think that there might be a chance that  the creation  of multiple interpolation instances (in a  loop)  consumes too much memory. This can be a flaw in the implementation. @heistermann can definitely say much more about this, if he is available again (couple of days).

Anyway, I can test and try to figure it out, but I won't have any time in the next few weeks.

itati01, 2016-07-27:
@kmuehlbauer Thank you very much so far and your time for testing! Here is my simple script (the missing part is all about setting up the netcdf). Just run it with a file pattern as argument, e.g. radolan2nc_test.py \*08\*bin\*

[radolan2nc_test.zip](https://github.com/wradlib/wradlib/files/387238/radolan2nc_test.zip)

itati01, 2016-07-28:
@kmuehlbauer 
Possibly an issue with scipy/numpy. Upgrading to scipy 0.18.0 and numpy 1.11.1 did not work. After copying ipol.py into my script and inserting `del` commands, memory consumption remained at 220 MB. 

```
def interpolate(src, trg, vals, Interpolator, *args, **kwargs):
    if vals.ndim == 1:
        ix_valid = np.where(np.isfinite(vals))[0]
        ip = Interpolator(src[ix_valid], trg, *args, **kwargs)
        result = ip(vals[ix_valid])
        del ip                                                         # changed
    elif vals.ndim == 2:
        ip = Interpolator(src, trg, *args, **kwargs)
        result = ip(vals)
        del ip                                                          # changed
        nan_in_result = np.where(np.isnan(result))
        for i in np.unique(nan_in_result[-1]):
            ix_good = np.where(np.isfinite(vals[..., i]))[0]
            ix_broken_targets = (nan_in_result[0]
                                 [np.where(nan_in_result[-1] == i)[0]])
            ip = Interpolator(src[ix_good],
                              trg[nan_in_result[0]
                              [np.where(nan_in_result[-1] == i)[0]]],
                              *args, **kwargs)
            tmp = ip(vals[ix_good, i].reshape((len(ix_good), -1)))
            del ip                                                       # changed
            result[ix_broken_targets, i] = tmp.ravel()
    else:
        # removed Exception here
        else:
            ip = Interpolator(src, trg, *args, **kwargs)
            result = ip(vals[ix_valid])
            del ip                                                       # changed
    return result

def interpolate_polar(data, mask=None, Interpolator=Nearest):
    # removed unchanged code
    filling = interpolate(src_coord, trg_coord, values_list, Interpolator)
    filled_data[clutter_indices] = filling.astype(filled_data.dtype)
    del trg_coord, filling                                        # changed
    if np.any(np.isnan(filled_data)):
        trg_coord = (np.array([binx[np.where(np.isnan(filled_data))],
                              biny[np.where(np.isnan(filled_data))]])
                     .transpose())
        filling = interpolate(src_coord, trg_coord, values_list,
                              Interpolator=Nearest)
        filled_data[np.where(np.isnan(filled_data))] = filling
        del trg_coord, filling                                   # changed
    del src_coord                                                 # changed
    return filled_data.reshape(data.shape[0], data.shape[1])
```

kmuehlbauer, 2016-07-28:
scipy 0.18.0 issue please see #86

itati01, 2016-07-28:
yes, i simply changed the import in util.py to numpy.nanmean

kmuehlbauer, 2016-07-28:
> After copying ipol.py into my script and inserting del commands, memory consumption remained at 220 MB.

Unfortunately I had no time to further look into this. But your testing seems to back my assumption about multiple instances of the `interpolator`.

kmuehlbauer, 2016-07-28:
> yes, i simply changed the import in util.py to numpy.nanmean

Yes, this solves the `nanmean` stuff. But I got some more errors regarding cKDTree implementation with 0.18.0 (#86). Are you not experiencing this?

kmuehlbauer, 2016-07-28:
For using interpolate_polar/interpolate in a loop, we need to rework this a little.

The instance ``` ip = Interpolator(src, trg, *args, **kwargs)``` can be re-used throughout the whole loop. Now it is re-created for every iteration, which might lead to the accumulation of memory. Normally the python garbage collector takes responsibility, don't know why this doesn't work correctly.

I'll test your code snippet locally and report back. Maybe it is an OS issue.

``` python
def interpolate(src, trg, vals, Interpolator, *args, **kwargs):
    if vals.ndim == 1:
        ix_valid = np.where(np.isfinite(vals))[0]
        ip = Interpolator(src[ix_valid], trg, *args, **kwargs) # can possibly be re-used
        result = ip(vals[ix_valid])
        del ip                                                         # changed
    elif vals.ndim == 2:
        ip = Interpolator(src, trg, *args, **kwargs)   # can be re-used
        result = ip(vals)
        del ip                                                          # changed
        nan_in_result = np.where(np.isnan(result))
        for i in np.unique(nan_in_result[-1]):
            ix_good = np.where(np.isfinite(vals[..., i]))[0]
            ix_broken_targets = (nan_in_result[0]
                                 [np.where(nan_in_result[-1] == i)[0]])
            ip = Interpolator(src[ix_good],
                              trg[nan_in_result[0]
                              [np.where(nan_in_result[-1] == i)[0]]],
                              *args, **kwargs)  # might not be re-usable
            tmp = ip(vals[ix_good, i].reshape((len(ix_good), -1)))
            del ip                                                       # changed
            result[ix_broken_targets, i] = tmp.ravel()
    else:
        # removed Exception here
        else:
            ip = Interpolator(src, trg, *args, **kwargs)  # can be re-used
            result = ip(vals[ix_valid])
            del ip                                                       # changed
    return result
```

itati01, 2016-07-28:
currently not, only a runtime warning in clutter.py (division by zero) for Apr 22, 2008 14:50 when subsetting and masking nodata before calling the clutter function. but the raster looks strange, anyway.

kmuehlbauer, 2016-07-28:
Your [test snippet](https://github.com/wradlib/wradlib/files/387238/radolan2nc_test.zip) is running here on 2015 RW data without problems or any memory accumulations (linux-64bit) with latest wradlib. Did you also test this stripped version without netcdf-code?

itati01, 2016-07-28:
i tested the stripped version earlier and am currently running it with the netcdf code. is a bit slow with the interpolation but it is also beyond the first year.

kmuehlbauer, 2016-07-28:
OK, the whole year is processed now. No problem whatsoever with it. So this is obviously a problem with windows or your special win32-system. 

I can't reproduce the error.

itati01, 2016-07-28:
ok, good to know. keep up the good work.

kmuehlbauer, 2016-07-29:
Closing this. Can't reproduce.

closed 2016-07-29 by kmuehlbauer