wradlib/wradlib #89: ImportError with wradlib 0.8.x (scipy.stats.nanmean is deprecated)

itati01, 2016-07-28:
Hello,

I cannot import wradlib 0.8.x with scipy 0.18.0 since "scipy.stats.nanmean is deprecated in scipy 0.15.0 in favour of numpy.nanmean" [http://docs.scipy.org/doc/scipy-0.17.0/reference/generated/scipy.stats.nanmean.html](http://docs.scipy.org/doc/scipy-0.17.0/reference/generated/scipy.stats.nanmean.html)

> Traceback (most recent call last):
>   File "..\radolan2nc_test1.py", line 5, in <module>
>     import wradlib
>   File "C:\Python27\lib\site-packages\wradlib\__init__.py", line 31, in <module>
> 
>     from . import adjust  # noqa
>   File "C:\Python27\lib\site-packages\wradlib\adjust.py", line 112, in <module>
>     import wradlib.ipol as ipol
>   File "C:\Python27\lib\site-packages\wradlib\ipol.py", line 43, in <module>
>     import wradlib.util as util
>   File "C:\Python27\lib\site-packages\wradlib\util.py", line 33, in <module>
>     from scipy.stats import nanmean
> ImportError: cannot import name nanmean

kmuehlbauer, 2016-07-28:
This will be resolved in upcoming version 0.9.0. Until then we recommend using scipy < 0.18.0.

kmuehlbauer, 2016-07-29:
fixed with this commit: 3dd57a195dcdb562672d058ce51b0ec6fa2961b8

closed 2016-07-29 by kmuehlbauer