wradlib/wradlib #98: Organizing slideshow cell peroperties for each notebook.

heistermann, 2016-08-26:
In the IPython notebook, you can set attributes to each cell the `Cell Toolbar`. Activating `View > Cell Toolbar > Slideshow`. After that, you can set, for each cell, its property in case the notebook is rendered as a slideshow. This can be done by

```shell

$ ipython nbconvert notebook.ipynb --to slides

```

or in case the html slides shoudl be directly served:

```shell

$ ipython nbconvert notebook.ipynb --to slides --post serve

```
Here's an [example](http://www.slideviper.oquanta.info/tutorial/slideshow_tutorial_slides.html#/).

We can set the slideshow attributes without affecting the remaining notebook functionality. This feature can be used to achieve a better presentability of the notebooks during courses/presentations. For this purpose, all cell properties need to be set in all notebooks. Sometimes this requires to rethink the internal organisation of the notebooks.

kmuehlbauer, 2016-08-28:
This is a really useful feature. I've used this recently to present a quick introduction to wradlib and stuff to colleagues. 

I'm totally open and in support to add this functionality to the wradlib notebooks. Can several notebooks be combined into one presentation? If not, this means some problems, because we would need to create really huge notebooks.

@heistermann Can you explain a bit more how this might be adapted for wradlib notebooks?

heistermann, 2016-08-29:
For the start, I'd just consider each notebook to be a seperate presentation.  `$ ipython nbconvert notebook.ipynb --to slides` renders this to one single bulk html slideshow which is then served together with the Reveal.js library. 

We could check whether `to-- slides` supports cross-referencing of notebooks. In that case, we could also use the TOC tree for specific topics and browse through it in a presentation. In the simplest case, though, you would just serve one specific html slideshow.

So for now, we would just need to set, for each cell of a notebook, the slideshow property: you can select between

- `slide`: starts a new slide
- `fragment`: adds a fragment to a slide
- `subslide`: vertically aligned slide which sets up an additional navigation path (subtopics/digression)
- `skip`: skip that
- `-`: do nothing (not sure how it differs from `skip`
- `note` shows a note for the presenter

Setting the properties can be an ongoing task - every time someone used a specific notebook as a slideshow, he or she can integrate that through a pull request. I'll set up an example pull request soon.

heistermann, 2016-08-29:
Wow, this is even better (someone already recommended it to me some months ago - was it you, Kai? - but I had not tried it, yet:

https://github.com/damianavila/RISE

You can easily install it via

```shell

$ source activate wradlib
$ conda install -c damianavila82 rise

```

That't it. Now you can enter the live reveal slideshow from inside the notebook (there's a button in the Toolbar) - no static rendering required! You can control the slideshow using the `slideshow` properties - as you would be using `nbconvert`. From the live slideshow, you can execute or edit cells - interactive slideshow. We have to do nothing execpt setting the slideshow prpoerties of the cells. we can navigate to other notebooks just as before.

kmuehlbauer, 2016-08-29:
@heistermann Yes, it was me, but I somehow lost track of RISE. 

If you want to have some changes in the css you can put a custum.css into .juypter/custom/. Then you can override fontsize etc. for better display.

heistermann, 2016-09-22:
Partly addressed in Pull Request #104. However, there are still notebooks left for which the slideshow properties need to be set.

kmuehlbauer, 2018-11-01:
@heistermann

Coming back to this, how about creating these slideshows to be available via wradlib.org website? Like my Talk Slides from last ERAD. Shouldn't be that difficult to implement.

kmuehlbauer, 2018-11-01:
We should move this Issue to wradlib-notebooks.