wradlib/wradlib #105: How to compute cumulative beam blockage

heistermann, 2016-09-21:
In our notebook `beamblockage/wradlib_beamblock.ipynb`, we compute the cumulative beamblockage fraction this way: find the maximum fraction along a beam, then set every bin *before* that to the beam bloackage fraction PBB, and everything *after* that to the maximum value. I think this is incorrect in cases where we have local maxima *before* the absolute maximum. Please have a look at the following image: Would you agree that it should look like the "new implementation"?

[image]

If you agree, I would change the notebook accordingly. I would also add a convenience function to `wraldib.qual` in order to compute the cumulative beam blockage fraction. I would add all this to Pull Request #104.

kmuehlbauer, 2016-09-21:
@heistermann Believe it or not, I'm working this morning with the beamblock notebook for incorporation in a radar lecture. Good catch! :+1: for fixing and convenience function.

kmuehlbauer, 2016-09-21:
@heistermann BTW, do you know or have seen a matplotlib based plot, where the coordinate system is curvilinear to see the earth's curvature? I'm trying to get this in for better representation.

heistermann, 2016-09-21:
Sorry, I don't.

closed 2016-09-22 by heistermann