wradlib/wradlib #111: Need help converting raw iris data to odim_h5

barisuzun, 2016-10-09:


IreneCrisologo, 2016-10-09:
Hi @barisuzun ,

Can you share some sample data? 
I have a prototype script for reading IRIS but it is for IRIS data exported as netcdf. I can try to see if the script can be used directly for your data, or if it has to be adapted a little bit.

barisuzun, 2016-10-09:
Thanks for your interest and kindness Irene.

[IRIS_MAX.zip](https://github.com/wradlib/wradlib/files/518340/IRIS_MAX.zip)

heistermann, 2016-10-14:
Dear @barisuzun,

thanks for this request. In general, we are hesitant with adding support for additional file formats in wradlib. However, let's see how Irene's attempt works out.

However, I see another alternative: Py-ART provides support for reading the IRIS (i.e. SIGMET) file format. There is also a way to map Py-ART data objects to BALTRAD data objects. BALTRAD, again, provides the opportunity to write data objects to ODIM_H5. You could try to work this out using the [open virtual machine for weather radar science](http://openradar.github.io) since both Py-ART and BALTRAD are available on that machine. In the [ERAD2016 short course](http://openradar.github.io/erad2016), Daniel Michelson had shown a notebook which exemplifies that bridge between Py-ART and BALTRAD. Maybe we can also work out a recipe that reads your sample data and then converts to ODIM_H5.

IreneCrisologo, 2016-10-25:
Hi @barisuzun,

I tried using Py-ART for reading the files you provided, but it does not work. The structure of the files does not seem to fit the expectation of the Py-ART sigment i/o function.

So I looked into the files themselves, and after some investigation and playing around with the data, I came up with this plot:

[image]

which looks like a Maximum-Intensity-Plot. 

The code to make that plot is here:

~~~~
import numpy as np
import pylab as pl
import os

datapath = 'D:/DATA/IRIS_MAX/URUN'
filename = 'IZM150807140052.MAXEWSF'

fn = os.path.join(datapath,filename)
f = open(fn, 'rb')
r = np.fromfile(f, dtype='B')
f.close()

data_array = r[640:]
data_array = np.reshape(data_array,(720,720))

pl.figure(figsize=(15,15))
pl.pcolormesh(data_array, cmap=pl.cm.spectral)
pl.xlim((0,720))
pl.ylim((0,720))
pl.show()
~~~~

The metadata of this dataset is still hidden within the first several hundred characters which cannot be decoded without the documentation. The values are between 0 and 255, so one cannot infer what unit it is in. There must be a way to convert it to the actual unit value. The data then starts from the 640th character. Reshaping the data to a square array and plotting it directly gives the image above, including the gray areas which signify the boundaries and empty spaces of the plot.

So it seems that the data is a processed product, in gridded form. It looks like the plot itself was exported as a binary file.

To proceed I think one should work with the raw data. For that I think Py-ART can be used. And as Maik said, proceed with combining the use of Py-ART and BALTRAD to create a ODIM_H5 file.

kmuehlbauer, 2017-03-07:
Closing. Please reopen, if appropriate.

kmuehlbauer, 2017-06-17:
@barisuzun

Sometimes things are already there, one only must find them. I remembered that gdal is able to read IRIS raster files.

I added an example to the wradlib-gists: https://github.com/wradlib/wradlib-gists/blob/master/formats/iris_raster/read_iris_max_product.ipynb

closed 2017-06-17 by kmuehlbauer