wradlib/wradlib #112: plot netcdf

godofyoga, 2016-10-17:
Hello ,
 I am having netcdf file of X-Band EEC radar and it consist 17 elevation angle with different radar parameters.
I able to read file using following code in wradlib.
import wradlib as wb
import pylab as pl
filename = '/home/user/20140421/y20140421/MDV-20140421-130257-PPIVol.nc'
netcdf_f =wb.io.read_generic_netcdf(filename)
print netcdf_f['variables']

I unable to plot date so how can i separate each elevation angle data and make plot?

Thank you.

wradlib, 2016-10-17:
Hi Yogesh,

in order to get an impression of the nc-file, could you attach the 
output of

`ncdump -h /home/user/20140421/y20140421/MDV-20140421-130257-PPIVol.nc`

It's also of interest, if this is just radardata within netcdf or 
complying to some standard (eg. cfradial).

Thanks,
Kai

Am 17.10.2016 um 14:03 schrieb yogesh kolte:
> Hello ,
>  I am having netcdf file of X-Band EEC radar and it consist 17 elevation angle with different radar parameters.
> I able to read file using following code in wradlib.
> import wradlib as wb [...]

kmuehlbauer, 2016-10-17:
@godofyoga 

In order to get an impression of the nc-file, could you attach the output of 

`ncdump -h /home/user/20140421/y20140421/MDV-20140421-130257-PPIVol.nc` 

It's also of interest, if this is just radardata within netcdf or complying to some standard (eg. cfradial).

godofyoga, 2016-10-17:
netcdf MDV-20140421-130257-PPIVol {
dimensions:
	time = 13896 ;
	range = 2500 ;
	sweep = 18 ;
	frequency = 1 ;
	string_length_sm = 10 ;
	string_length_md = 25 ;
variables:
	int volume_number ;
		volume_number:standard_name = "data_volume_index_number" ;
		volume_number:units = "1" ;
	double latitude ;
		latitude:standard_name = "latitude" ;
		latitude:units = "degrees_north" ;
	double longitude ;
		longitude:standard_name = "longitude" ;
		longitude:units = "degrees_east" ;
	double altitude ;
		altitude:standard_name = "altitude" ;
		altitude:units = "meters" ;
	char time_coverage_start(string_length_md) ;
		time_coverage_start:standard_name = "data_volume_start_time_utc" ;
		time_coverage_start:units = "1" ;
	char time_coverage_end(string_length_md) ;
		time_coverage_end:standard_name = "data_volume_end_time_utc" ;
		time_coverage_end:units = "1" ;
	double time(time) ;
		time:standard_name = "time" ;
		time:long_name = "time in seconds since volume start" ;
		time:units = "seconds since 2014-04-21T13:02:57Z" ;
	float azimuth(time) ;
		azimuth:standard_name = "beam_azimuth_angle" ;
		azimuth:long_name = "azimuth_angle_from_true_north" ;
		azimuth:units = "degrees" ;
		azimuth:axis = "radial_azimuth_coordinate" ;
	float elevation(time) ;
		elevation:standard_name = "beam_elevation_angle" ;
		elevation:long_name = "elevation_angle_from_horizontal_plane" ;
		elevation:units = "degrees" ;
		elevation:axis = "radial_elevation_coordinate" ;
	float radar_antenna_gain_h ;
		radar_antenna_gain_h:meta_group = "radar_parameters" ;
		radar_antenna_gain_h:standard_name = "nominal_radar_antenna_gain_h_channel" ;
		radar_antenna_gain_h:units = "dB" ;
	float radar_antenna_gain_v ;
		radar_antenna_gain_v:meta_group = "radar_parameters" ;
		radar_antenna_gain_v:standard_name = "nominal_radar_antenna_gain_v_channel" ;
		radar_antenna_gain_v:units = "dB" ;
	float radar_beam_width_h ;
		radar_beam_width_h:meta_group = "radar_parameters" ;
		radar_beam_width_h:standard_name = "half_power_radar_beam_width_h_channel" ;
		radar_beam_width_h:units = "degrees" ;
	float radar_beam_width_v ;
		radar_beam_width_v:meta_group = "radar_parameters" ;
		radar_beam_width_v:standard_name = "half_power_radar_beam_width_v_channel" ;
		radar_beam_width_v:units = "degrees" ;
	float radar_receiver_bandwidth ;
		radar_receiver_bandwidth:meta_group = "radar_parameters" ;
		radar_receiver_bandwidth:standard_name = "radar_receiver_bandwidth" ;
		radar_receiver_bandwidth:units = "s-1" ;
	float frequency(frequency) ;
		frequency:meta_group = "instrument_parameters" ;
		frequency:standard_name = "radiation_frequency" ;
		frequency:units = "s-1" ;
	float radar_measured_transmit_power_h(time) ;
		radar_measured_transmit_power_h:meta_group = "radar_parameters" ;
		radar_measured_transmit_power_h:standard_name = "radar_transmit_power_h_channel" ;
		radar_measured_transmit_power_h:units = "watts" ;
	float radar_measured_transmit_power_v(time) ;
[... 157 lines cut ...]
		CCORH:units = "dB" ;
		CCORH:_FillValue = -32768s ;
		CCORH:scale_factor = 0.01f ;
		CCORH:add_offset = 0.f ;
		CCORH:coordinates = "elevation azimuth range" ;
	short CCORV(time, range) ;
		CCORV:long_name = "clutter_correction_v" ;
		CCORV:units = "dB" ;
		CCORV:_FillValue = -32768s ;
		CCORV:scale_factor = 0.01f ;
		CCORV:add_offset = 0.f ;
		CCORV:coordinates = "elevation azimuth range" ;

// global attributes:
		:Conventions = "Cf/Radial instrument_parameters radar_parameters" ;
		:title = "PPIVol" ;
		:institution = "EEC" ;
		:references = "EEC" ;
		:source = "EDGE" ;
		:history = "original" ;
		:comment = "PPIVol" ;
		:instrument_name = "MDV" ;
		:site_name = "MDV" ;
		:n_gates_vary = "false" ;
}

kmuehlbauer, 2016-10-17:
It's cfradial (See the Conventions attribute). My suggestion would be to use `pyart` to read the data.

godofyoga, 2016-10-18:
I am using Py-Art also but how can i use radar object of pyart in wradlib and make plot.
I want to use attenuation correction in wradlib.

kmuehlbauer, 2016-10-18:
You can't use the part radar object out of the box. But you can extract the needed features/moments from the radar object as bumpy arrays and use them within wradlib. 

Let me know, if you need more help with this extraction. I can come back to this early next week.

kmuehlbauer, 2016-10-25:
@godofyoga Is this still an issue? Have you been successful in extracting the needed data?

godofyoga, 2016-10-26:
I able to plot data with py-Art but not with wradlib. I want to use attenuation correction algorithms in wradlib.

Please help me

kmuehlbauer, 2016-10-26:
Can you share a minimal plotting example for `py-art`. Then I can better assist you in extracting the wanted data from the nc-file.

Also, sharing an example data file would be nice. You can send all to the mailinglist (wradlib-users@googlegroups.com).

kmuehlbauer, 2017-03-07:
resolved by [this post](https://groups.google.com/d/msg/wradlib-users/5Vgdn5vtVMg/PcuMd_oRCgAJ) on the wradlib user forum.

closed 2017-03-07 by kmuehlbauer