wradlib/wradlib #113: Read RQ and FZ products

UbyX, 2016-10-18:
Hello,
reading  RQ and FZ files fails currently, because some header tokens are missing.
updating def get_radolan_header_token() like this helps:

    head = {'BY': None, 'VS': None, 'SW': None, 'PR': None,
            'INT': None, 'GP': None, 'MF': None, 'VV': None, 'QN': None, 
            'MS': None, 'LV': None, 'CS': None, 'MX': None, 'BG': None}

Since I can not test other products, I will leave this suggestion here.
Are there reasons, that not all are included? If not, maybe you can implement it after testing.

Regards

kmuehlbauer, 2016-10-18:
@UbyX 

Which wradlib version are you using? FZ product reading capability was introduced in wradlib with PR #68 and is available in `wradlib 0.9.0`. The two new header token ('MF' and 'VV') have been introduced with commit https://github.com/wradlib/wradlib/pull/68/commits/b37188ddae85eddd56b0430bb4eb692c3540bbc8.

There is the header token 'QN' in your list. This comes from RQ/RE-product, but is not crucial to further processing (so we didn't incorporate it). If there is some use case, we can implement this token too.

UbyX, 2016-10-18:
@kmuehlbauer 

I use an older version. I looked into the master code and did not see 'QN'...
On a second look, the rest is there. Now I feel silly ;)

closed 2016-10-18 by UbyX

kmuehlbauer, 2016-10-18:
@UbyX No worries, we've all been there & done that. :grinning: