wradlib/wradlib #116: Implement procedure for 3-D matching of TRMM/GPM radars with ground radars

heistermann, 2016-10-26:
Rob Warren from Monash University has implemented the procedure according to Schwaller and Morris (2011) in IDL. Given his consent, we should explore ways to port this implementation to wradlib, making sure that existing wradlib infrastructure is used as extensively as possible.

First step would be to share the code and provide a minial example including sample data.

Schwaller, MR, and Morris, KR. 2011. A ground validation network for the Global Precipitation Measurement mission. J. Atmos. Oceanic Technol., 28, 301-319.

heistermann, 2016-11-03:
Based on the code and data sent by Rob, we might want to develop a task list in order to avoid redundant efforts. My naive suggestion would be to break down the code into functions - where this is not yet the case - and then think about which functions can be leveraged by native wradlib functions. 

However, we might also need to re-organise the entire control flow in order to make it more generic. Here, we have to think about which parts should be incorporated into wradlib modules and which should be left to scripting/application programming/recipes.

The main procedure is contained in `prgr_matchvol.pro` and we need to start out from there. Porting the other code files - reading data, date string conversion, cropping etc. - appears trivial. However, we need to decide how to represent the different data sources.

This is the basic (roughly simplified) flow in `prgr_matchvol.pro`:

```
lots of pre-settings
loop over a defined set of days:
   search for matching swaths that overpassed on a day
   loop over each of these swaths:
      read the data for one swath
      georeference the PR bins (x,y,z + bin volume)
      filter for "real" precipitation
      reflectivity conversion (S/C/Ku)
      read and time-match GR scans
      georeference GR bins (x,y,z + bin volume)
      ...
      create lots of containers to store intermediate results
      loop over each PR profile of a swath:
         loop over each GR elevation:
            and HERE (ll. 661-774) comes the actual matching/averaging 
      some minor post-processing
      save results as netcdf
      create routine plots	  

```

My feeling is that we should start out from one specific GPM/TRMM swath/orbit and a matching ground radar scan. All the time-matching and time loops could be wrapped around this later. Based on this, we could develop a task list. This is just a first order proposal!

### Read and georeference PR data
- [ ] `def`` read a PR (GPM or TRMM) swath file
- [ ] `def`` compute x,y,z coordinates of an arbitray bin in a PR profile
- [ ] `def`` correct PR x,y,z for parallax (?)

### Read and georeference GR data 
- [ ] `def`` approximate the volume of an arbitray bin in a PR profile
- [ ] `def`` read a GR sweep (based on wradlib)
- [ ] `def`` compute x,y,z coordinates of an arbitray bin in a GR sweep (based on wradlib)
- [ ] `def`` approximate the volume of an arbitray bin in a GR sweep

### Routine conversions
- [ ] `def`` reflectivity conversion between bandwidths
- [ ] `def`` dither the reflectivities (?)

### Matching and averaging: find a smart way to break down ll. 661-774
- [ ] `def` match PR and GR bins
- [ ] `def` average GR reflectivity (lienar units) over PR bin volume

I think the filtering (e.g. use only precipitating bins) can be applied along the lines (before or after).

Where should we put this? What about already moving the sample data into the wradlib/wradlib-data repo, and to create a feature branch in which we build this as an example first, and then move indvidual functions into wradlib module?

heistermann, 2016-11-03:
@RobWarren: Is it ok if we put the sample data into our [wradlib/wradlib-data](https://github.com/wradlib/wradlib-data) repository?

kmuehlbauer, 2016-11-03:
@heistermann It would be a great idea to put the example data into our  [wradlib/wradlib-data](https://github.com/wradlib/wradlib-data) repository.

For the approximation of the volume of one arbitrary bin, we have [wradlib.qual.pulse_volume()](https://github.com/wradlib/wradlib/blob/master/wradlib/qual.py#L119) which might be used for starters.

IreneCrisologo, 2016-11-03:
@heistermann i agree with all your suggestions. Where can one see the code?

heistermann, 2016-11-03:
@IreneCrisologo Here's the [IDL code and the sample data](https://github.com/wradlib/wradlib/files/569424/volmatch.tar.gz). The data will be in wradlib/wradlib-data soon, and I'll reference a branch with a notebook in which we can put together the different components.

heistermann, 2016-11-03:
I've created a new development branch in my fork [heistermann/wradlib/match_gpm](https://github.com/heistermann/wradlib/tree/match_gpm). It contains a [notebook](https://github.com/heistermann/wradlib/blob/match_gpm/notebooks/match3d/wradlib_match_trmm_gpm.ipynb) which only reads the data so far.

@kmuehlbauer What would be the appropriate way to work together on this? Would you need to fork from my fork or should we replicate the new branch in wradlib/wradlib? I could also move this elsewhere...

IreneCrisologo, 2016-11-03:
Thanks for this Maik. I'll be following the thread to see how I can contribute as well. This should be my first exercise in forks and branches (better late than never..)

Will you try to run this in IDL first or just jump directly in re-writing in Python?

kmuehlbauer, 2016-11-03:
@heistermann Yes, we can work on the branch in your fork for the time being. I'll just add your fork as another remote and pull your dev-branch into my local repo. Then I can push changes to my fork and initiate a pull request to your branch. @IreneCrisologo this would also be a way to dive into git with forks and branches.

Concerning IDL, good question. I would volunteer to test converted python code within my python/idlbridge/IDL-environment. So we can test python functions against their IDl-counterpart. 

I already run the IDL-code and wrote a quick plotting mimic for the output hdf. I'll add a more complete visualization later. 

``` python
import wradlib as wrl
import matplotlib.pyplot as pl
import numpy as np
import os

def discrete_cmap(N, base_cmap=None):
    """Create an N-bin discrete colormap from the specified input map"""

    # Note that if base_cmap is a string or None, you can simply do
    #    return plt.cm.get_cmap(base_cmap, N)
    # The following works for string, None, or a colormap instance:

    base = pl.cm.get_cmap(base_cmap)
    color_list = base(np.linspace(0.1, 1, N))
    cmap_name = base.name + str(N)
    return base.from_list(cmap_name, color_list, N)

fpath = '/local/kai/progs/volmatch/'
fpath1 = os.path.join(fpath, 'GPM_example')
fname1 = os.path.join(fpath1, '20141206094829_004383.nc')

outdict = wrl.io.read_generic_netcdf(fname1)
print(outdict.keys())

nrejgr = outdict['variables']['nrejgr']['data']
nrejpr = outdict['variables']['nrejpr']['data']
ntotgr = outdict['variables']['ntotgr']['data']
ntotpr = outdict['variables']['ntotpr']['data']
frac1 = nrejpr/ntotpr
frac2 = nrejgr/ntotgr
ipair = (frac1 <= 0.1) & (frac2 <= 0.1)
print(ipair)

refgr = outdict['variables']['refgr']['data'][ipair]
x = outdict['variables']['x']['data'][ipair] / 1000.
y = outdict['variables']['y']['data'][ipair] / 1000.
ds = outdict['variables']['ds']['data'][ipair] / 1000.
itilt = outdict['variables']['itilt']['data'][ipair]
print(refgr.shape, x.shape, y.shape, ds.shape)

tilt_max = np.argmax(np.bincount(np.asarray(itilt, dtype=np.int64)))
print(tilt_max)

ipairm = (itilt == tilt_max)
refgr = refgr[ipairm]
x = x[ipairm]
y = y[ipairm]
ds = ds[ipairm]

clabel = 'Reflectivity [dBZ]'
pl.scatter(x, y, s=ds*10., c=refgr, cmap=discrete_cmap(12, 'jet'), vmin=0, vmax=60)
pl.gca().set_aspect('equal')
cbar = pl.colorbar(ticks=np.arange(0, 60, 5), pad=0.05, label=clabel)
pl.clim(0, 60)
pl.gca().set_xlim(-150, 150)
pl.gca().set_ylim(-150, 150)
pl.show()
```

RobWarren, 2016-11-04:
Wow this is progressing very quickly. Sorry for not being more responsive; I just have a ridiculous amount on at the moment. It sounds like the code hasn't been too hard to follow, which is good.

@heistermann Yes feel free to add the sample data to your repository.

I'd be curious to know how you guys plan on going about the actual volume matching. As detailed in [this conference proceeding](https://ams.confex.com/ams/34Radar/techprogram/paper_155254.htm), the original algorithm uses (1) a simple linear average over PR bins between the top and bottom of the radar beam half width at the point of intersection and (2) an inverse distance weighted average of the GR bins within the radius of the PR beam at the point of intersection. 

Bob Morris informed me via email that they later added linear volume weighting of the GR bins to account for the fact that close to the radar the bin volume can vary significantly within the PR footprint. I included this in my code but it probably isn't relevant as I exclude samples at short ranges; specifically, less than 15 km which for a 1deg beam width corresponds to a beam thickness of less than 250m, the gate spacing of TRMM.

Bob also explained the reason why the PR bins are averaged linearly rather than using distance weighting: "The reason for the linear averaging of the PR/DPR gates is because the vertical gradients of the PR reflectivity are small over the extent of the GR half-power points over which they are averaged, and secondly, because the true height of the GR beam is not always certain due to non-standard refraction."

I should point out one important difference between my code and the original algorithm. The latter applies a minimum threshold of 15 dBZ on the GR reflectivity, whereas I use 0 dBZ. I found that using the higher threshold led to underestimation of a negative GR bias at low reflectivities. Morris himself noted (both in his correspondence with me and [this conference proceeding](https://ams.confex.com/ams/35Radar/webprogram/Paper191729.html)) that the GR-PR difference you obtain is very sensitive to this choice of threshold. The best way to deal with the issue is to compute the difference using an iterative procedure as outlined by [Protat et al. (2011)](http://journals.ametsoc.org/doi/abs/10.1175/2010JTECHA1443.1); this is what I did to correct my GR reflectivities.

kmuehlbauer, 2016-11-07:
@heistermann @IreneCrisologo @RobWarren 

See PR https://github.com/heistermann/wradlib/pull/1

Added first two submodules (io and vis). Put `read_gpm`-function in `io_func` and `comparison`-function in `vis_func`.

Also added the traitsui graphical dictionary editor for quick and easy dictionary examination.

kmuehlbauer, 2016-11-07:
FYI: The TRMM-data is in hdf4-format which can't be read by h5py-based readers. I'm currently implementing a solution using `gdal`.

kmuehlbauer, 2016-11-07:
> The TRMM-data is in hdf4-format which can't be read by h5py-based readers. I'm currently implementing a solution using gdal.

:tada: We can do it with gdal, but it is a bunch of lines. Even easier is using `netcdf4`. Although it is hdf4, we can use the `generic_netcdf`-reader.

kmuehlbauer, 2016-11-08:
See PR https://github.com/heistermann/wradlib/pull/2

I'll add a `read_gr`-function later today. Then we have everything in place to start the actual code conversion.

kmuehlbauer, 2016-11-10:
Just fancying around with github features: https://github.com/wradlib/wradlib/projects

heistermann, 2016-12-02:
If you like, you could have a brief look at [this notebook](https://github.com/heistermann/wradlib/blob/match_gpm/notebooks/match3d/wradlib_match_workflow.ipynb). It tries to pull through with the complete matching procedure for one single elevation angle of the ground radar. You can set the elevation angle index `ee` at the very beginning.

At least the preliminary results look somewhat resonable. It still needs a lot of code smoothing, formalisation, modularisation etc. Beyond, we can start to play around with different approaches of how to compute the volume average, how to filter the data, whether to average over linear units etc.

kmuehlbauer, 2016-12-02:
@heistermann Congratulations for taking those last steps. This looks really nice!

RobWarren, 2016-12-04:
@heistermann Looks good.

I spotted what I think is a typo in the section "Read and organize the data" - after the second elif, should it not be `platf=="trmm2a25":`?

A few other points...

**Averaging in linear vs log units**

I do all the averaging in linear units. My argument is that we are trying to compare what the two radars "see" and Z is directly proportional to the received power; however, I'm not sure whether that stands up. I consistently find that I get a strong positive trend in GR reflectivity offset when the standard deviation (SD) of GR reflectivity in the averaging volume increases above ~8 dB. Thus when the sample volume contains a wide range of GR reflectivity values, averaging of these values generates a positive bias compared to TRMM/GPM. This is illustrated in the figure below which shows the GR offset as a function of GR Reflectivity SD for four radars in the Australian network. The black diamonds and vertical lines show the median and interquartile range, respectively, while the full underlying distribution is indicated in the colours. Open circles show the number of comparisons (overpass events) for each SD bin, with the scale given on the right y axis.

[image]

**Platforms**

For TRMM, 2A23 and 2A25 are different _products_ not different platforms. You need both when comparing a GR with TRMM. The 2A23 product has the brightband and precipitation type data while the 2A25 product has the (attenuation-corrected) reflectivity.

**Brightband height**

In the comments for the section "Convert S band GR reflectivities to Ku band" you note the possibility of using the brightband height and width for each profile individually. The problem with this is that only stratiform profiles have a brightband. That is why I use the previously computed median (not mean) brightband height for all profiles within the domain.

**Pulse volume calculation**

My use of the small angle approximation in calculating the pulse volume was actually a coding error; I have since updated it in my code. So definitely use your in-built function.

**Rejected bins**

When averaging the reflectivities, it is important to store the number of bins below the sensitivity of the instrument. For TRMM this is 18 dBZ while for GPM it is a little lower (around 12 dBZ, I believe), although thus far I have been using 18 dBZ for both. For the GR I use a value of 0 dBZ. The ratio of these numbers to the total number of bins in the averaging volume then gives an indication of the degree of partial beam filling. Schwaller and Morris (2011) suggest excluding samples where the missing (rejected) fraction exceeds 0.3, but I use a more stringent value of 0.1.

heistermann, 2016-12-05:
@RobWarren Thank you so much for your in-depth comments! It will be really exciting to follow up on these thoughts. Just very briefly:

#### The effect of sample variance on offset estimation

The results you show are really interesting. I agree that averaging over linear units appears like the right choice. The dependence of the offset on standard deviation or variance is, as you showed, at the same time a dependence on the number of samples. Could the bias also be a sample size artefact of the offset estimation? I'm not good at statistics, so that's just an uneducated guess.

#### Platforms

Thanks for this clarification! I was aware that both variants are just different products, but not that we basically need both for this procedure.

#### Brightband height

This sounds very reasonable. I guess our task will be to make this as generic as possible so that any assumptions on bright band height (even from other sensors/methods such as MRR etc.) could be used while using the satellite product as default.

#### Rejected bins

I see - I guess it generally makes sense to store as much "procedural" information as possible for ex-post analyses. I even thought that we might write the actual bin values to file (not only the averages) in order to experiment with averaging and the like *after* the actual matching.

RobWarren, 2016-12-06:
@heistermann Re- the effect of sample variance on offset estimation, I should clarify: by sample size I mean the number of PR-GR volume-matched pairs rather than the number of GR bins averaged within that volume. There does seem to be a separate dependence on the latter; namely, an increasing trend in GR offset with number of GR bins averaged, although this is not true for all radars. This also shows up as a negative trend in GR offset with range, as illustrated in the figure below. Here I have plotted the median GR offset (for the same four sites as above) as a function of range and height (with respect to the brightband). Data are only shown where the sample size exceeds 100 and I have excluded samples with a GR SD > 8 dB. Both Mt Staplyton and Wollongong show a clear decreasing trend with range at all levels, but for Newcastle and Terry Hills the pattern is less clear. The fact the range dependence is not universal makes me wonder if it is related to elevation pointing errors, although non-standard refraction may also be an issue. If it were related to the process of averaging over GR bins then we would expect to see a consistent trend for all sites. Again, I welcome your thoughts on this.

[image]

heistermann, 2016-12-07:
After I managed to read the TRMM data with a new function `io_func.read_trmm_gdal`, I pushed the  [the same notebook](https://github.com/heistermann/wradlib/blob/match_gpm/notebooks/match3d/wradlib_match_workflow.ipynb), but with the results for the TRMM case. Please have a look, if you like.

heistermann, 2016-12-07:
For clarity: here are permanent links to the notebooks with TRMM example and to the GPM example, so the linked notebooks won't change with each new commit.

- TRMM example: [link to notebook at commit 513580f](https://github.com/heistermann/wradlib/blob/531580f4ebff3c4d9eaf3c74cebed17c6bcf7c61/notebooks/match3d/wradlib_match_workflow.ipynb)
- GPM example:  [link to notebook at commit 6534aea](https://github.com/heistermann/wradlib/blob/6534aea6c34fc9764d4197375fa86c3693b5c288/notebooks/match3d/wradlib_match_workflow.ipynb)

heistermann, 2017-01-10:
Should we start to think about which functions of this procedure should be ported to wradlib modules? We would have to answer the following questions:

- Which functions (or objects) should be ported?
- Where should they be ported (which module)?
- Which changes might be required to allow the move (change name, design, break down into multiple functions etc.)?

Right now, new functions have been put to `io_func.py` and `external.py`. We might find that we want to define more new functions. In that case, I suggest to put them to `external.py` first, and then think about how to integrate to wradlib.

Following is the current list of functions to think about (check the box once the function has been ported).

**From external.py:**

- [ ] `pr_pars` (a dict): not move at all?
- [ ] `correct_parallax`: move to `wradlib.georef`? 
- [ ] `sat2pol`: move to `wradlib.georef`?
- [ ] `dist_from_orbit`: move to `wradlib.georef`?
- [ ] `calculate_polynomial`: move to `wradlib.util`?
- [ ] `get_bb_ratio`: move to `wradlib.qual`?
- [ ] `ku2s` (a dict): move to `wradlib.trafo`?
- [ ] `fix_for_cband`: move to `wradlib.trafo`?
- [ ] `s_ku_coefficients`: move to `wradlib.trafo`?

**From io_func.py:**

These are basically all functions that create a data model, based on Rob's suggestions. This data model is required to make the current code work. I think we should aim at getting rid of this data model in the long run. I would not suggest to move it wradlib. If at all, we should use a more general data model such as the one based on CfRadial/ODIM_H5. To allow for a smooth transition, we might want to keep the functions from `io_func.py` where they are?

In addition, we shoudl see how to organise further parts of the workflow as functions. Just add suggestions here:

- [ ] ...

kmuehlbauer, 2017-01-17:
@heistermann Sorry, didn't had time so far looking into this. Will have a look this week.

kmuehlbauer, 2017-02-16:
@heistermann and me worked a bit on the final implementation.

We transferred most functions into wradlib-codebase in [this feature branch](https://github.com/wradlib/wradlib/tree/gpm_to_wrl). An updated notebook is also [there](https://github.com/wradlib/wradlib/blob/gpm_to_wrl/notebooks/match3d/wradlib_match_workflow.ipynb). A cleanup follows and then the final merge into `master`.

kmuehlbauer, 2017-02-20:
@heistermann @IreneCrisologo @RobWarren

I have updated the [feature branch](https://github.com/wradlib/wradlib/tree/gpm_to_wrl). The preliminary docs are [online](http://wradlib.org/wradlib-docs/latest/notebooks/match3d/wradlib_match_workflow.html). @heistermann will polish it for the final merge into `master`. @RobWarren you might also have another look or two at it.

IreneCrisologo, 2017-03-07:
Hi all,

I'm trying to distinguish between convective rain and stratiform rain and in the PR dataset, this is given by `pr_type = pr_data['ptype']`. I want to assign each matched pair a `pr_type`, and hopefully this will reveal if those points that cause a spread in the matching are coming from the convective rains. 

Do you have any suggestion how to assign a `pr_type` to a matched pair considering that for each matched volume both PR and GR are averaged? 

Thank you!

IreneCrisologo, 2017-03-07:
Or did I have a wrong understanding of the dataset?

The precipitation type is a 2D array while the reflectivity data is a 3D array.
[image]

Which means I can just select the `pr_type` based on the coordinates, like this?
`type_pr = pr_type[ss, rr]`

Or do I also have to flip the array upside down?

Thanks again.

kmuehlbauer, 2017-03-07:
@IreneCrisologo If I'm not mistaken, then you should be able to get the precipitation type per beam like you already suggested. No flipping needed, AFAIK.

IreneCrisologo, 2017-03-07:
Alright, great! Thanks @kmuehlbauer !

kmuehlbauer, 2017-03-23:
This is now in the code base. Closing. Please open new issues for specific problems.

closed 2017-03-23 by kmuehlbauer