wradlib/wradlib #117: kdp_from_phidp sobel vs. convolution

kmuehlbauer, 2016-10-28:
We have the mentioned  two kdp_from_phidp functions (`kdp_from_phidp_sobel`, `kdp_from_phidp_convolution`) which are obviously doing the same. 

`kdp_from_phidp_convolution` has two advantages:

- using vectorized `convolve1d` on the whole array
- calculate derivation filter only once

Both functions are using convolution with the same kernel in two different ways.  `kdp_from_phidp_sobel` convolutes the kernel with the data and handles the edge cases by handcrafted mirroring at the edge.  `kdp_from_phidp_convolution` convolutes the data with the reversed kernel and uses the standard 'reflect' mode for edge cases.

Further difference in handling of the edge cases: `kdp_from_phidp_sobel` handles ray start and end section in any case via linear regression, where `kdp_from_phidp_convolution` handles this only when invalid data (nan) is calculated.

The kernel used in both functions is most likely a derivative of an interpolation filter. This way we get the function derivative with just one convolution. 

The kernel values can be calculated by this function:

y = (12 * x) / (L^3 * L)

where L is the kernel width and x runs from  -(L - 1.) / 2 to  (L - 1.) / 2 with steps of 1.

The python function would look something like this:

```
def kernel(L):
    N = (L - 1.) / 2.
    x = np.linspace(-N, N, L)
    window = x * 12. / (np.power(L, 3.) - L)
    return window
```

It does make sense to consolidate the two functions into one. We can also think about adding a bit more flexibility to the convolution part in terms of edge case handling and also interpolation and derivation. Just think about phidp data with different noise overlay (diff. frequency) etc. It would come in handy to create kernels with different interpolation schemes.

Looking forward to the discussion.

ping @scollis who posted the sobel-filter on SO

heistermann, 2016-10-28:
I totally agree! The variety of available functions originates from a time (three years ago) when we our development policy was less strict ;-) - maybe you remember when you wrote 

> [...] so schnell kann man ja gar nicht gucken, wie die PHIDP/KDP processing enhancements rauskommen.

Maybe a good start would be to set up some test cases with synthetic data - particularly with regard to noise level, gaps, and edge cases -  which we could use to compare behaviour, and then also directly use for our unit tests. Certainly, it would be good to harmonize and clean up the API.

Please note that right now, *none* of these filters is used in wradlib except `process_raw_phidp_vulpiani` except `kdp_from_phidp_convolution`. From that perspective, cleaingin up will not be much of a problem.

kmuehlbauer, 2016-11-03:
@heistermann

After taking another math lesson I've assembled everything in a little [jupyter notebook gist](https://gist.github.com/kmuehlbauer/efdcf063d8254d7007ffef77940ab25e).

It turned out that the convolution is just a means to do the summation of the Lanczos Differentiator. My best suggestion would be to name everything appropriately (also add/rename functions) and leave references to the used math in the code.

kmuehlbauer, 2018-03-14:
resolved with #241

closed 2018-03-15 by kmuehlbauer