wradlib/wradlib #120: [3D-Matching]: Georeferencing GR/PR

kmuehlbauer, 2016-11-10:
Looking at the IDL-source the GR projection is setup with some transversal mercator projection using the radar center coordinates. This could be done within python with wradlib-georef functions:

```python
wgs84 = wradlib.georef.get_default_projection()
lon0 = gr_data['lon']
lat0 = gr_data['lat']
alt0 = gr_data['alt']
er1 = wradlib.georef.get_earth_radius(lat0, wgs84) *4./3.
a = wgs84.GetSemiMajor()
b = wgs84.GetSemiMinor()

rad = wradlib.georef.proj4_to_osr(('+proj=aeqd +lon_0={lon:f} ' + 
                                   '+lat_0={lat:f} +a={a:f} ' +
                                   '+b={b:f}').format(lon=lon0,
                                                       lat=lat0,
                                                       a=a, b=b))
er2 = wradlib.georef.get_earth_radius(lat0, rad)*4./3.
print(er1, e2)
```
The GR grid in the `aeqd`-projection is created like this:

```python
# create range and azimuth arrays
rmax = gr_data['r0'] + gr_data['ngate'] * gr_data['dr']
rng = np.arange(0, gr_data['ngate'][0]) * gr_data['dr'][0] + gr_data['dr'][0]/2.
azi = np.arange(0, gr_data['nbeam'][0]) - gr_data['a0'][0]
# create gr lonlat grid
# use lowest elevation (first int his case)
el = gr_data['elang'][0]
gr_grid = np.meshgrid(rng, azi)
rlon, rlat, ralt = wradlib.georef.polar2lonlatalt_n(gr_grid[0], gr_grid[1], el, (lon0, lat0, alt0 ))
gr_ll = np.dstack((rlon, rlat))
gr_xy = wradlib.georef.reproject(gr_ll, projection_source=wgs84, projection_target=rad)
```
The PR grid:

```python
lonp = pr_data['lon']
latp = pr_data['lat']
pr_x, pr_y = wradlib.georef.reproject(lonp, latp, projection_source=wgs84, projection_target=rad)
pr_xy = np.dstack((pr_x, pr_y))
```

heistermann, 2016-11-10:
Absolutely!

kmuehlbauer, 2016-11-17:
@heistermann 

It might be necessary that we split `wradlib.georef.polar2lonlatalt_n()` into two functions. One function just creates the x,y,z-coordinates (name: `polar2xyz()`). The function should be able to take refractivity into account.  `polar2lonlatalt_n` would just call this function inside. 

Reason is, it doesn't make sense to calculate x,y,z, then transform it to lon, lat and reproject it again to x,y,z.

We also need to think about how to setup the `aeqd`-projection inside `polar2lonlatalt_n`. ATM we assuming a spherical earth with equal semimajor and semiminor axis. This might introduce unwanted coordinate differences.

Finally we also need  the inverse of those functions, to calculate (range, azimuth, elevation) of arbitrary PR bins.

kmuehlbauer, 2016-11-24:
Assuming the constellation like the image below is showing, we have the following:

[image]

```python
def sat2pol(prcoords, grcoords, re):
    
    # calculate arc length
    s = np.sqrt(np.sum(prcoords[...,0:2]**2, axis=-1))
    
    # calculate arc angle
    gamma = s / re
    
    # calculate theta (elevation-angle)
    numer = np.cos(gamma) - (earth_radius+rcoords[2])/(earth_radius+scoords[...,2])
    denom = np.sin(gamma)
    theta = np.rad2deg(np.arctan(numer/denom))
    r = (earth_radius+scoords[...,2])*denom/np.cos(np.deg2rad(theta))
    phi = 90 - np.rad2deg(np.arctan2(scoords[..., 1], scoords[..., 0]))
    phi[phi <= 0] += 360
    return r, theta, phi
```

See also this [gist](https://gist.github.com/kmuehlbauer/32092e1ccac66608156c77d772246464) for more equations.

@RobWarren Can you have a look, if I got everything right?

RobWarren, 2016-11-24:
@kmuehlbauer 

Should 'earth_radius' not be replaced with 're'? Presumably, re is the _effective_ Earth radius? Seeing your previous comment, am I right in saying that wradlib has the functionality to account for non-standard refraction?

I imagine that the errors introduced by assuming a spherical earth are pretty small compared to the footprint of the PR (~5 km) over the ranges being considered here (<150 km) so you're probably safe to stick with that (i.e. use a spherical map projection rather than the Transverse Mercator).

kmuehlbauer, 2016-11-25:
@RobWarren Yes, it should read `re` aka effective Earth radius. I changed it accordingly. 

In wradlib we are using the functions from Doviak (beam height, arc distance), which account for earth curvature and refraction. Those functions have `k` as parameter, which defaults to 4/3. So there is room for some (linear) non-standard refraction.

kmuehlbauer, 2017-03-07:
see #129

We might need to open a new issue to solve this issue:

_"It might be necessary that we split `wradlib.georef.polar2lonlatalt_n` into two functions. One function just creates the x,y,z-coordinates (name: polar2xyz()). The function should be able to take refractivity into account. polar2lonlatalt_n would just call this function inside.

Reason is, it doesn't make sense to calculate x,y,z, then transform it to lon, lat and reproject it again to x,y,z."_

closed 2017-03-07 by kmuehlbauer