wradlib/wradlib #121: [3D-Matching]: GR/PR indices match-up

kmuehlbauer, 2016-11-10:
For the calculation of PR-Grids within the GR radar domain I suggest using `ZonalDataPoint` and `GridPointsToPoly`.

```python
gr_domain = gr_xy[:,-1,:]
gr_domain = np.vstack((gr_domain, gr_domain[0]))
l_gr = []
l_gr.append(gr_domain)

zd = wradlib.zonalstats.ZonalDataPoint(pr_xy.reshape(-1, pr_xy.shape[-1]), l_gr, srs=rad, buf=0.)
obj1 = wradlib.zonalstats.GridPointsToPoly(zd)
```

As the PR-Data is only 176x49 this is outstanding fast. And we could think about the possibility to add attributes to the points in the object. We also might think about creating polygonic footprints from the PR-Data and do the same to find matching GR-bins.

In the attached image you can see which points are considered with the method in IDL (simple bounds) and using the zonal-approach. 

[image]

heistermann, 2016-11-10:
This is exactly what I was hoping for - to make some use of `zonalstats` for the 2-D components of the analysis.

kmuehlbauer, 2017-03-07:
see #129

closed 2017-03-07 by kmuehlbauer