wradlib/wradlib #123: [3D-Matching]: PR/GR Pulse Volume

kmuehlbauer, 2016-11-14:
The original IDL-code uses [Small-angle approximation](https://en.wikipedia.org/wiki/Small-angle_approximation) to calculate the tangent of `theta` (equals radius of the beam cone frustum). Also it assumes the cone frustum being a cylinder, which simplifies more.

The wradlib-function [wradlib.qual.pulse_volume()](https://github.com/wradlib/wradlib/blob/master/wradlib/qual.py#L119) uses this cylinder simplification too, but doesn't use the small-angle approximation. 

@heistermann If I'm not mistaken the wradlib function is missing a factor of 0.5 in the tangent-angle calculation. Correcting this, we can use the `pulse_volume()`-function to calculate pulse volume of PR/GR pulses.

heistermann, 2016-11-15:
@kmuehlbauer dammit you're right. We need to use half of the aperture angle to get the base radius of the cone (or cylinder). Well, this bug was long asleep...should I push an immediate fix?

heistermann, 2016-11-15:
See Pull Request #124.

kmuehlbauer, 2016-11-15:
Merged #124

closed 2017-03-07 by kmuehlbauer