wradlib/wradlib #125: Zonal Stats - Curved shape of mask misses some parts of original shapefile

IreneCrisologo, 2016-12-02:
I encountered some issues with the Zonal stats when i used a polar grid input.

But first here is a screenshot from Qgis when i put all the generated shapefiles (trg, src, dst) together.
trg is blue - the original shapefile
src is orange - based on masked used to reduce the grid size
dst is purple - final shapefile used to calculate zonal stats

[image]

The src shapefile is based on the mask, which is based on the bounding box of trg plus some buffer. However, here it seems that when the bounding box is made, the corners are joined following the polar space, so that it curves. And because of the curve it misses some parts of the original shapefile (trg). The resulting shapefile for calculation (dst) is then also missing the lower parts of the original shapefile.

kmuehlbauer, 2016-12-02:
@IreneCrisologo How did you create the mask? Did you call `mask_from_bbox` with parameter `polar=True`?

IreneCrisologo, 2016-12-02:
@kmuehlbauer  yes, as in the notebook example.

IreneCrisologo, 2016-12-02:
To adjust the bounding box so that the mask fits my shapefile, I moved the lower left point of the bounding box like so:

```python
bbox_adj = dict(left=bbox[0] + buffer*2.5, right=bbox[1] + buffer,
            bottom=bbox[2] - buffer*2, top=bbox[3] + buffer)
```

by moving the lower left point to the right and downwards. The bounding box is marked by the black dot for the lower-left and upper-right points.

Before:
[image]

After:
[image]

kmuehlbauer, 2016-12-02:
@IreneCrisologo It seems, that the `polar`-algorithm doesn't correctly detect/find the range-indices. The upper part is no problem here, but the lower.  We also need to be careful with the ranges not only with the angles.

kmuehlbauer, 2016-12-02:
@IreneCrisologo I can confirm, that the problem is in calculation of the range min index (near radar index). The range max index is not effected. Need to think about this.

kmuehlbauer, 2016-12-05:
@IreneCrisologo 

I did not check for speed, but this is the brute force method for the calculation of the indices within the bounding box. 

It finds the x-ccords between left and right and the y-coords between top and bottom. The two boolean output fields are also then with `&` to get a boolean field marking indices within the bounding box as true. 

```python
mask = (((coords[..., 1] > bbox['bottom']) &
         (coords[..., 1] < bbox['top'])) &
        ((coords[..., 0] > bbox['left']) &
         (coords[..., 0] < bbox['right'])))
```

Can you check this within your notebook. It works for some artificial box here.

@heistermann We might just use this simple construct in the notebook. It's reasonable fast. I can't remember why we did the `cKDTree` in the first place, but adding the `polar` keyword made the function quite complex. And, do we need this `shape` return at all?

IreneCrisologo, 2016-12-05:
@kmuehlbauer  it works great! 

[image]

This is so simple and it seems like it would work regardless if the data is polar or not.

kmuehlbauer, 2016-12-05:
> This is so simple and it seems like it would work regardless if the data is polar or not.

Yes, this will work for any type of grid-structure. Glad, that this works for you.

heistermann, 2016-12-05:
@kmuehlbauer Hm, I *think* we used the original approach in order to make sure to mask a regular 2-D grid (polar or Cartesian) instead of a collection of points. This way, we could e.g. plot the masked subset using `pcolormesh`. 

However, from the `zonalstats` perspective, this is not a required functionality. The above approach for subsetting is way more intuitive and appears to work for polar and Cartesian grid types. 

Now the question is how to proceed. Options?
- [ ] replace the current version of `wradlib.zonalstats.mask_from_bbox` with the new approach;
- [ ] leave the function as is, and just exemplify the new approach in the notebook (new function not required?)
- [ ]  drop `wradlib.zonalstats.mask_from_bbox` from the API, and just exemplify the new approach in the notebook

Other options?

kmuehlbauer, 2016-12-05:
@heistermann That was my gut feeling too, but as you already pointed out, it is not needed for `zonalstats`.

I would tend to No. 2 of the options. This would also mean to fix `wradlib.zonalstats.mask_from_bbox` for polar grids. However, we can also consider option 3, if we are reasonably sure that such functionality is not needed.

heistermann, 2016-12-05:
We might be able to fix this issue more easily once we introduced a function to `wradlib.georef` which converts from geographic (or Cartesian) coordinates to polar (or spherical) coordinates (as part of issue #120).

kmuehlbauer, 2018-02-14:
@heistermann With `xyz_to_spherical` available in #215 , how do we proceed here?

kmuehlbauer, 2018-03-29:
resolved by https://github.com/wradlib/wradlib-notebooks/pull/8

closed 2018-03-29 by kmuehlbauer