wradlib/wradlib #126: Read hdf4 files with netcdf4 issue.

heistermann, 2016-12-05:
Read the TRMM hdf4 files with netCDF4 fails:

```
nc.Dataset(trmm_2a23_file, 'r')
...
netCDF4\_netCDF4.pyx in netCDF4._netCDF4.Dataset.__init__ (netCDF4\_netCDF4.c:12617)()

IOError: NetCDF: Unknown file format
```

Same with

```
nc.Dataset(trmm_2a23_file, 'r', disk_format="HDF4")
```

Is this a Windows issue with linking the correct libraries and stuff (see e.g. [here](https://github.com/Unidata/netcdf4-python/issues/581)?

heistermann, 2016-12-05:
On [quintusdias/hdfeos_python_zoo](https://github.com/quintusdias/hdfeos_python_zoo), it is said that  

> [...] the netcdf4 library was not compiled with hdf4 support [on Anaconda].

Not sure how this holds for conda-forge. The `hdf4` library is available in my environment, but I guess this does not necessarily imply that hdf4 support is available...

The above repo suggests to use `gdal` for reading hdf4 - as @kmuehlbauer already did.

kmuehlbauer, 2016-12-05:
@heistermann Have you tried to update all relevant conda packages? It might have been introduced lately. If this doesn't work, we would need to take the 'gdal'-way. Unfortunately I threw away the code I already did :-(

heistermann, 2016-12-05:
@kmuehlbauer yes, I created a new environment and made a fresh install of netcdf4 (version 1.2.5) which seems to be the latest version.

With gdal, I am able to open the dataset:

```
>>> from osgeo import gdal
>>> hdf = gdal.Open(trmm_2a23_file)
>>> sdsdict = hdf.GetMetadata()
>>> for key in sdsdict.keys():
>>>    print key
InputRecord
FileHeader
FileInfo
SwathHeader
NavigationRecord
JAXAInfo
```

I guess now the tedious part begins...

kmuehlbauer, 2016-12-05:
> yes, I created a new environment and made a fresh install of netcdf4 (version 1.2.5) which seems to be the latest version. No success.

The hdf4 is compiled in `libnetcdf`, but if you created a fresh environment, so this should have pulled also the latest `libnetcdf `.

Yes the extraction of the data will take some lines of code...

heistermann, 2016-12-06:
When I read the metadata of the TRMM hdf4 files using gdal, I get a bunch of attributes that I cannot really match with the attributes you have been using in `io_func.read_trmm` (such as `Year`, `Month` etc.).

The following code is an example to extract the metadata:
```
hdf = gdal.Open(trmmfile)
meta = hdf.GetMetadata_Dict()
for key in meta.keys():
    print ""
    print(key)
    attrs = meta[key].split()
        for attr in attrs:
            print(attr)
```

For the 2A23 file, this gets me:

```
InputRecord
InputFileNames=1C21.20100206.69662.7.HDF;
InputAlgorithmVersions=7.53;
InputGenerationDateTimes=2011-08-13T12:31:04.000Z;

FileHeader
AlgorithmID=2A23RW;
AlgorithmVersion=7.12;
FileName=2A23.20100206.69662.7.HDF.ps.hdf;
GenerationDateTime=2016-08-06T05:19:54.000Z;
StartGranuleDateTime=2010-02-06T11:14:22.114Z;
StopGranuleDateTime=2010-02-06T11:15:19.660Z;
GranuleNumber=69662;
NumberOfSwaths=1;
NumberOfGrids=0;
GranuleStart=SOUTHERNMOST_LATITUDE;
TimeInterval=ORBIT;
ProcessingSystem=PPS;
ProductVersion=7;
MissingData=0;

FileInfo
DataFormatVersion=m;
TKCodeBuildVersion=1;
MetadataVersion=m;
FormatPackage=HDF
Version
4.2
Release
7,
February
6,
2012;
BlueprintFilename=TRMM.V7.2A23.blueprint.xml;
BlueprintVersion=BV_13;
TKIOVersion=3.70.8;
MetadataStyle=PVL;
EndianType=LITTLE_ENDIAN;

SwathHeader
NumberScansInSet=1;
MaximumNumberScansTotal=10000;
NumberScansBeforeGranule=0;
NumberScansGranule=97;
NumberScansAfterGranule=0;
NumberPixels=49;
ScanType=CROSSTRACK;

NavigationRecord
LongitudeOfMaximumLatitude=23.169094;
SolarBetaAngleAtBeginningOfGranule=-51.191399;
SolarBetaAngleAtEndOfGranule=-51.180939;

JAXAInfo
CalibrationCoefficientVersion=2;
GranuleFirstScanUTCDate=2010/02/06;
GranuleFirstScanUTCTime=09:51:31;
GranuleFirstScanUTCMilliseconds=862;
GranuleLastScanUTCDate=2010/02/06;
GranuleLastScanUTCTime=11:23:55;
GranuleLastScanUTCMilliseconds=499;
SoftwareVersion=3;
DatabaseVersion=5;
TotalQualityCode=G;
LongitudeOnEquator=-60.924435;
UTCDateOnEquator=2010/02/06;
UTCTimeOnEquator=10:14:38;
UTCMillisecondsOnEquator=136;
CenterScanUTCDate=2010/02/06;
CenterScanUTCTime=10:37:43;
CenterScanUTCMilliseconds=081;
FirstScanLat=-35.116158;
FirstScanLon=-144.970612;
LastScanLat=-35.113762;
LastScanLon=-168.597687;
NumberOfRainScans=2040;
```

How do these correspond with the attributes extracted in `io_func.read_trmm`?

Extracting the actual variables (Latitude, Longitude, rainType, rainFlag etc.) was not a problem.

kmuehlbauer, 2016-12-06:
This is just the metadata. The actual datasets can be retrieved like this:

```python
subds = hdf.GetSubDatasets()
for ds in subds:
    print(ds[1], ds[0])
```

Extracting Latitude:
```python
dataset = wradlib.io.open_raster( subds[0][0])
lat = wradlib.georef.read_gdal_values(dataset)
print(lat.shape)
```

There might be other solutions, but this seems the fastest. Unfortunately we need to do the mapping somehow.

heistermann, 2016-12-06:
Yes, I know... sorry for the misunderstanding. This is not about the subdatasets, but about the additional metadata. In `io_func.read_trmm`, you do e.g.

`year = pr_data1['variables']['Year']['data']`,

but `Year` is **not** a subdataset. How do I access this variable? E.g., in 2A23, `.GetSubDatasets()` only returns subdatasets for `Latitude, Longitude, rainFlag, rainType, status, HBB, BBwidth`. What about `Year, Month` etc.?

kmuehlbauer, 2016-12-06:
It's there, but hidden. I now remember, that only the subdatasets you showed are claimed by `GetSubDatasets()`.

The filestring is looking like this:

`'HDF4_SDS:UNKNOWN:"/local/kai/python/projects/wradlib-data/trmm/2A-RW-BRS.TRMM.PR.2A23.20100206-S111422-E111519.069662.7.HDF":9'`

The `9` after the colon is actually the dataset number. But `GetSubDatasets()` only gets the dataset 9 to 15. So you can read the other datasets by replacing the number after the colon by 0 to 8 to get the other datasets. Those 9 datasets are `Year`, `Month`, `DayofMonth`, `Hour`, `Minute`, `Second`, `MilliSecond`, `DayofYear`, `scanTime_sec` (as of the netcdf reader names).

heistermann, 2016-12-06:
Wow, great. Thanks! But how do I know which of the number from 0-8 are `Year, Month, DayofMonth, Hour, Minute, Second, MilliSecond, DayofYear, scanTime_sec`? They do not have a description string like the other subdatasets...

kmuehlbauer, 2016-12-06:
No idea how to tackle this. I think, that this is specified/standardized somewhere. Since the nectdf-reader is mapping to some reasonable name, this should be inside the file somewhere. Gdal seems not to be able to get hold of those names.

heistermann, 2016-12-06:
I searched for other ways, but to no avail. For now, I guess we have to trust that the order of indices/names of those hidden subdatasets is same for each TRMM hdf4 file...

heistermann, 2016-12-07:
We can avoid some level of ambiguity by testing the "units" keyword of the metadata of each subdataset (which will return "years", "months" etc.). However, there are two *hidden* subdatasets with units "days" and two with units "s" (=seconds).

kmuehlbauer, 2016-12-07:
Then, we can further differentiate "days" by comparing the first element. The greater one is `DayofYear`, the lesser one `DayOfMonth`. If they are equal, it's January 😬.

The "s" is integer in the case of `Second` and floating point in the case of `scanTime_sec`.

heistermann, 2016-12-07:
Ain't them great: self-describing formats!

heistermann, 2016-12-07:
Ok, I've added a new function `io_func.read_trmm_gdal` - it appears to work, but it is certainly not the desired solution as compared to usign netcdf4...

kmuehlbauer, 2016-12-07:
@heistermann I opened an issue at conda-forge: https://github.com/conda-forge/libnetcdf-feedstock/issues/11. Think this will get sorted out in the near future.

kmuehlbauer, 2016-12-08:
@heistermann This is resolved with https://github.com/conda-forge/libnetcdf-feedstock/pull/12. I just tested this under win7. It's working as expected.

closed 2016-12-08 by kmuehlbauer

heistermann, 2016-12-08:
Thank you so much for making this happen! Too bad about the new function `read_trmm_gdal`, though :cry:

kmuehlbauer, 2016-12-08:
At least we gained experience in using gdal/companions 😬.