wradlib/wradlib #135: raster data dimensions and origin

kmuehlbauer, 2017-03-19:
Lately I was processing some raster data and found it useful to facilitate some of the wradlib `georef` and `io` functions. Doing this I found some inconsistencies in the use of dimensions ((rows, cols) vs. (cols, rows)) and origin ('upper' vs. 'lower').

rows, cols:
- `io.read_RADOLAN_composite`, origin=lower
- `georef.get_radolan_grid`, origin=lower
- `georef.read_gdal_coordinates`, origin=upper, geotiff/raster standard
- `georef.read_gdal_values`, origin=upper, geotiff/raster standard
- `io.read_raster_data`, origin=upper -> lower (via np.flipud)
- `io.read_Rainbow`, origin=?, for pixmap-data
- `io.read_safnwc`, origin=upper

cols, rows:
- `io.to_AAIGrid`, origin=upper ?
- `io.to_GeoTIFF`, origin=lower, BUG: dimensions swapped, breaks when Xdim!=Ydim

Obviously wradlib is using (rows, cols) with origin='lower' in some readers, but (rows, cols) origin='upper' in others. Georeferenced Raster Formats like GeoTIFF and others use (rows, cols) origin='upper' too. The two writers (`io.to_GeoTIFF` at least) are broken, with a dimension mismatch.

The main question: **To which nomenclature do we stick?** 

My suggestion would be to use *(rows, cols)* in any case and origin *'lower'* or *'upper'* when appropriate. To achieve that, we would need to adapt the functions. This could be achieved by using keyword *origin* within the reader and writer functions. Although *origin='upper'* is widely used in imaging (e.g. GeoTIFF) and stuff, *origin='lower'* is also used widely in map coordinate systems.

Suggestions?

kmuehlbauer, 2017-03-21:
OK, I made some progress. Assuming we have something like this:

xy - coords array (rows, cols, 2), origin='upper'
data - (rows, cols), origin='upper'
srs - osr-projection-object

This creates a minimal In-Memory Raster of the given data:

```python
# create destination in-memory raster
mem_drv = gdal.GetDriverByName('MEM')
x_ps, y_ps = xy[1,1] - xy[0,0]
geotran = [xy[0, 0, 0], x_ps, 0, xy[0, 0, 1], 0 , y_ps]
rows, cols = data.shape
ds = mem_drv.Create('', cols, rows, 1, gdal.GDT_Float32) 
ds.SetGeoTransform(geotran)
ds.SetProjection(src_srs.ExportToWkt())
ds.GetRasterBand(1).SetNoDataValue(np.nan)
ds.GetRasterBand(1).WriteArray(data)
```
Now, we can do everything we want with this `ds`-dataset. Resampling, Reprojection, Raster-Output.

To GeoTIFF:

```python
target = gdal.GetDriverByName('GTiff').CreateCopy(filename, ds, 0)
del target
```

To AAIGrid (this creates also the prj-file):
```python
target = gdal.GetDriverByName('AAIGrid').CreateCopy(filename, ds, 0)
del target
```

and many, many more.

My suggestion is to have only ONE dedicated raster writing function, eg. `write_raster_dataset`.
We would then need one function to create a raster_dataset from data-array (and -type), coordinate-array and projection. 

Ideas?

heistermann, 2017-03-21:
Sounds very reasonable! I guess that based on the above snippet, functions to create a regular (GDAL) raster dataset (`create_raster_dataset`), and to write such a regular dataset to file (`write_raster_dataset`) would be quite straightforward to implement (at least for you ;-)). I agree that it makes sense to seperate *creation* from *writing* as some users might want to do other stuff with such a dataset - as you already pointed out (reprojection, resampling).

It is quite an awesome perspective as it would basically allow users to just provide a raster format code to `create_raster_dataset` (see http://www.gdal.org/formats_list.html) and this way create virtually any raster format, most importantly including netcdf and hdf5. 

So I guess the question is: How to deal with `io.to_AAIGrid` and `io.to_GeoTIFF`? Should we keep both as wrapper functions to provide backward compatibility? Or should we just drop them with a DeprecationWarning? I'd prefer the latter.

The other thing, which I do not yet fully understand: How do we deal with the different standards in the different readers - as you pointed out in your inital post? Do we keep these as they are, and only deal with the inconsistencies in case a future `create_raster_dataset` is used? In that case, the user would have to pass along this information (like "origin") so the transformation can be correctly handled? This means that the users have to take a good look at the docs (and we'd have to respecify the docs in that regard). But I think it is the only way, right? Because doing a transformation of the origin *inside* the reader function will probably break a lot of applications.

Thank you so much for pointing this issue out.

kmuehlbauer, 2017-03-22:
@heistermann 

I followed the suggestion for the two functions (`create_raster_dataset` and `write_raster_dataset`). See #137. Deprecation of both `to_AAIGrid` and `to_GeoTIFF` is very much agreed. I updated the [notebook](https://github.com/kmuehlbauer/wradlib/blob/raster-handling/notebooks/fileio/wradlib_gis_export_example.ipynb). 

There I have a small snippet of code which checks for the origin and flips the data and coords when necessary. That snippet could be used in some function `raster_origin` to automagically adapt the data and coords:

```python
def raster_origin(data, coords, direction):
    x_sp, y_sp = coords[1, 1] - coords[0, 0]
    origin = 'lower' if y_sp > 0 else 'upper'
    same = origin == direction
    if !same:
        data = np.flipud(data)
        xy = np.flipud(xy) + [0, y_sp]
    return data, coords

data, coords = raster_origin(data, coords, 'lower')
```

kmuehlbauer, 2017-03-22:
@heistermann 

Most of the functionality is already added. I think I will also deprecate the `io.read_raster_data`, because it is just two lines more of user code and we keep the magic to a minimum.

There is the `georef.resample_raster_dataset`-function which can resample the dataset before we extract the values and coordinates. This way we can use the interpolation functions of `gdal`. This can be extended by a concurrent reprojection (in the same call to `gdal.ReprojectImage()`) from source srs to some target srs. I have working code for this combination of resampling and reprojection. Since the  `georef.resample_raster_dataset` isn't used in the codebase, we can think about a change of names. Maybe `reproject_raster_dataset` is then a bit more coherent. The resampling would be just an addon per keywords.

Finally, we have some raster code in zonalstats, which can be merged with the new georef/io code. But that needs some refactoring. 

Stay tuned!

kmuehlbauer, 2017-03-23:
Refactoring is done, everything seems to work well. Moving to unit-tests.

kmuehlbauer, 2017-03-31:
resolved by #137

closed 2017-03-31 by kmuehlbauer