wradlib/wradlib #136: Calculate Vertically integrated liquid and Echo Tops?

lovechang1986, 2017-03-21:


kmuehlbauer, 2017-08-30:
@lovechang1986 Could you be a bit more specific what your source data is and what special requirements you might have for VIL and ECHOTOPS?

kuchaale, 2018-02-28:
[Possible source](https://vlab.ncep.noaa.gov/web/wdtd/-/vertically-integrated-liquid-vil-?selectedFolder=562123) how to calculate VIL.

pandasambit15, 2020-07-08:
Hi!

Even I am interested in this utility, if it is being worked at. Vertically Integrated Liquid (VIL), ECHO TOPS (ET) and Contoured Frequency Altitude Diagram (CFAD) are essential diagnostics for studying the vertical structure of precipitation and storms. 

Reference for CFAD: https://www.arm.gov/capabilities/vaps/radarcfad-121
Reference for VIL: https://vlab.ncep.noaa.gov/web/wdtd/-/vertically-integrated-liquid-vil-
Reference for ET: https://www.lakeeriewx.com/CaseStudies/EchoTops/EchoTops.html

Regards,
Sambit

syedhamidali, 2021-11-26:
You can check out Lakshmann et. al., 2013 for calculating echo top heights in the link provided below;
    [Link](https://doi.org/10.1175/WAF-D-12-00084.1)

kmuehlbauer, 2021-11-26:
@syedhamidali Thanks for resurrecting this rather old issue.