wradlib/wradlib #141: `testrunner.py` fails on OSX due to forking

cchwala, 2017-03-31:
`python testrunner.py -a` of the newest wradlib fails on OSX with the following output

```
$ python testrunner.py -a
wradlib.tests.test_adjust
wradlib.tests.test_atten
wradlib.tests.test_clutter
wradlib.tests.test_comp
wradlib.tests.test_dp
wradlib.tests.test_fill
wradlib.tests.test_georef
wradlib.tests.test_io
wradlib.tests.test_ipol
wradlib.tests.test_qual
wradlib.tests.test_trafo
wradlib.tests.test_util
wradlib.tests.test_verify
wradlib.tests.test_vis
wradlib.tests.test_vpr
wradlib.tests.test_zonalstats
wradlib.tests.test_zr

----------------------------------------------------------------------
Ran 0 tests in 0.000s

OK
runTest (__main__.NotebookTest) ... ['match3d', 'io_func']
ERROR

======================================================================
ERROR: runTest (__main__.NotebookTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "testrunner.py", line 62, in runTest
    self.assertTrue(__import__(self.name))
ImportError: No module named match3d.io_func

----------------------------------------------------------------------
Ran 1 test in 0.002s

FAILED (errors=1)
correctAttenuationHJ (wradlib.atten)
Doctest: wradlib.atten.correctAttenuationHJ ... ok
kdp_from_phidp_convolution (wradlib.dp)
Doctest: wradlib.dp.kdp_from_phidp_convolution ... The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
```

This is probably caused by #139

kmuehlbauer, 2017-03-31:
This is due to the fact, that we use TestSuites in subprocesses, introduced as said in #139.

This [stackoverflow issue](http://stackoverflow.com/questions/16254191/python-rpy2-and-matplotlib-conflict-when-using-multiprocessing) has some insight to this particular problem on OSX (and probably) windows.

We might consider checking travis-ci environment variables and use subprocesses only within the CI, not locally. Or we introduce a switch (--use-subprocess) to enable subprocesses. This seems to me more reasonable.

kmuehlbauer, 2017-03-31:
@cchwala You might test use snippet fore the time being.

@heistermann Didn't thought about specifics like that OSX issue. I think with the commandline-switch we are on the safe side, to use whatever is appropriate.

kmuehlbauer, 2017-03-31:
@cchwala This is fixed now with #142. 

If you want to use subprocess: 
```python testrunner.py -a -s```
and if not:
```python testrunner.py -a```.

commit 50e3571a (2017-03-31): FIX: add subprocess-switch in testrunner.py (#142)

* FIX: add subprocess-switch in testrunner.py

closes #141
closed 2017-03-31 by kmuehlbauer