wradlib/wradlib #145: ZeroDivisionError when PR has no overlap with GR data in matching workflow

IreneCrisologo, 2017-04-21:
In the step where the precipitation mask is generated (under **Subset relevant PR data**), the line `precip_mask = (pflag == 2) & wradlib.zonalstats.get_clip_mask(pr_xy, gr_domain, rad)`  produces a `ZeroDivisionError` because there is no overlap between the overpass array and ground radar data. 

I'm not sure if this should be caught somehow in the workflow, but i'm just reporting it in case someone encounters the same issue and cannot figure out the reason why the error occurs. It took some time to figure out, for me at least.

kmuehlbauer, 2018-02-13:
@IreneCrisologo 

This is actually a problem of the underlying ZonalData object. If there is no intersection the `dst` layer is empty. This leads to this faulty behaviour.

How should we tackle this? @heistermann We could add a property `has_intersections` (or similar) and test this before starting calculation.

kmuehlbauer, 2018-02-13:
Current workflow:

When creating the `ZonalDataPoint` or `ZonalDataPoly`, we assume that there will be intersections. This leads to the creation of a empty `dst` DataSource (DataSource with empty Layer). This is never tested in the subsequent processing.

Proposal:
We add a property `intersection_count` to `ZonalDataBase` and add the count there at creation time. We only create `GridPointsToPoly` or `GridCellsToPoly` if `intersection_count` is greater than 0.

kmuehlbauer, 2018-02-14:
fixed in #215

closed 2018-02-14 by kmuehlbauer