wradlib/wradlib #153: Missing "QN" flag in wradlib.io.get_radolan_header_token()

doktormerlin, 2017-06-13:
There is an issue in wradlib.io which prevents RADOLAN and RADVOR files with set QN flag to be read.

This is easily fixed by adding the "QN" flag to head dictionary in `wradlib.io.get_radolan_header_token()`

Without this flag, the end points from other header parts are read wrong, resulting in ValueErrors. 

The flag also has to be added to `wradlib.io.parse_DWD_quant_composite_header(header)` to further work with it in the output data of `wradlib.io.read_RADOLAN_composite(f, missing=-9999, loaddata=True)`

doktormerlin, 2017-06-13:
I'm on wradlib version 0.10.0

I am not sure if I am allowed to share the files, since they are not public. The RADVOR RQ product has the QN flag set, leading to `"ValueError: invalid literal for int() with base 10: '00000002QN 001'"`, with `k='MF', v=(74, 89)`, while `k='MF', v=(74, 83)` would be correct.

I will ask around if I am allowed to share an RQ file with you

kmuehlbauer, 2017-06-13:
Do you know which RADOLAN files have QN besides RQ? We have quite some samples in wradlib-data. But they all read fine.

kmuehlbauer, 2017-06-13:
You also might send just the header.

doktormerlin, 2017-06-13:
I don't know about any other RADOLAN Files having the QN header, it might be a header only used for RADVOR, which are forecasts generated by RADOLAN files and stored in the same format as RADOLAN.

A sample header for an RQ Product is `'RQ210945100000517BY1620162VS 2SW    1.7.2PR E-01INT  60GP 900x 900VV   0MF 00000002QN 001MS 67<bln,drs,eis,emd,ess,fbg,fld,fra,ham,han,muc,neu,nhb,ros,tur,umd>  '`

kmuehlbauer, 2017-06-13:
I think the header will be enough to check this. 

Your proposed change will fix this. Any chance to issue an PullRequest, @doktormerlin?

kmuehlbauer, 2017-06-14:
@doktormerlin Thanks for reporting this. The issue will be fixed with #154. Your provided header is correctly extracted with those changes.

PR #154 (2017-06-14): FIX: add 'QN' header token to RADOLAN reader
Add 'QN' header token to RADOLAN reader

closes #153
closed 2017-06-14 by kmuehlbauer