wradlib/wradlib #160: WRADLIB Release 1.0

kmuehlbauer, 2017-06-23:
The move to wradlib release 1.0 is covered inside this [project](https://github.com/wradlib/wradlib/projects/2).

This issue is the overall discussion starting point. New issues can be created to discuss specific aspects of this project.

kmuehlbauer, 2017-06-26:
IMHO it would be favourable to keep the changes of the several PR's to a minimum. See the differences between #155 and #159. In the latter only the first step (creation of submodules) whereas in the first all necessary changes  are applied. This lead to some unmanageable PR monster. To overcome this I would split the one big PR in severeal sub-PR's. 

@heistermann If there is consensus, I'll proceed with this suggested approach.

egouden, 2017-08-01:
Hi all,

First I would like to thank all contributors of wradlib.
I am still using wradlib for both research and production so I am happy to see that a release 1.0 is coming.
Unfortunately I have no time to contribute to the project at the moment.

I still have a few remarks :

- Is the meaning of version 1.0 obvious to everyone. Is it the same as this site http://semver.org/ ?
- Do you plan to offer compatibility with common production environment.
  For example RHEL 7 ships with numpy 1.7 but 1.10 is recommended by wradlib and some functionality will fail (e.g. read_gdal_values).

BR,

Edouard

kmuehlbauer, 2017-08-14:
> I still have a few remarks :
> 
>     Is the meaning of version 1.0 obvious to everyone. Is it the same as this site http://semver.org/ ?

In principle: YES. There are some last changes needed (#159), but the API should be stable then.

>     Do you plan to offer compatibility with common production environment.
>     For example RHEL 7 ships with numpy 1.7 but 1.10 is recommended by wradlib and some functionality will fail (e.g. read_gdal_values).

ATM we use `conda-forge` conda-environments for building and testing `wradlib`, but we have no numpy-matrix for that. So `wradlib` is always tested against the latest `numpy` (1.13 as of now). If you use `conda-forge` you get the whole (compatible) wradlib-stack installed for either python 2.7, 3.5 or 3.6.

kmuehlbauer, 2018-04-03:
closed with b5577d6c5797b787d4683dccb720e327c629113c

closed 2018-04-03 by kmuehlbauer