wradlib/wradlib #167: Get Reflectivity data from OPERA h5df

diegocarrio, 2017-07-19:
Dear community,

I am newbie with wradlib and I am experiencing some problems to obtain the data I am interested in. My work is related with data assimilation technique to improve weather forecasts. For such purpose, one of the objectives is to assimilate radar observations.
I have a h5df file from OPERA chart (http://eumetnet.eu/activities/observations-programme/current-activities/opera/) and what I need to be able to assimilate this data is to get the following parameters for each grid point:

- latitude
- longitude
- altitude
-  reflectivity

I have successfully obtained the latitude, longitude and altitude using the following code:

import numpy as np
import wradlib.georef as georef
import wradlib.io as io
import wradlib.util as util
#import warnings
#warnings.filterwarnings('ignore')
import matplotlib.pyplot as plt
import xarray as xr
import dcc

filename = './2012-10-14-18-00-00.itgat.PVOL.h5'
filename = util.get_wradlib_data_file(filename)
pvol = io.read_OPERA_hdf5(filename)

ntilt = 1
for i in range(100):
    try:
        pvol["dataset%d/what" % ntilt]
        ntilt += 1
    except Exception:
        ntilt -= 1
        break

nrays = int(pvol["dataset1/where"]["nrays"])
nbins = int(pvol["dataset1/where"]["nbins"])
rscale = int(pvol["dataset1/where"]["rscale"])
coord = np.empty((ntilt, nrays, nbins, 3))

for t in range(ntilt):
    elangle = pvol["dataset%d/where" % (t + 1)]["elangle"]
    coord[t, ...] = georef.sweep_centroids(nrays, rscale, nbins, elangle)

sitecoords = (pvol["where"]["lon"], pvol["where"]["lat"],
              pvol["where"]["height"])

proj_radar = georef.create_osr("aeqd", lat_0=pvol["where"]["lat"],
                               lon_0=pvol["where"]["lon"])
                               
radius = georef.get_earth_radius(pvol["where"]["lat"], proj_radar)

lon, lat, height = georef.polar2lonlatalt_n(coord[..., 0],
                                            np.degrees(coord[..., 1]),
                                            coord[..., 2], sitecoords,
                                            re=radius, ke=4. / 3.)

Dimension of these parameters are  (zdim=6, ydim=375, xdim=490). This data contains 6 elevation angles.

My question is: How could I obtain the values of reflectivity data for each elevation angle? Is there any function that could perform that?

Thanks in advance,
Diego Carrio

kmuehlbauer, 2017-07-19:
@diegocarrio First, glad you found the issue-tracker. Second, you can make code output beautiful by using backticks like this (already applied to your post):

    ``` python
    print("something")
    ```
You can extract the reflectivities from the pvol-dict-object. It includes all data, which is inside the odim-file. You can use your existing program with slight changes:

``` python
coord = np.empty((ntilt, nrays, nbins, 3))
relf = []
for t in range(ntilt):
    elangle = pvol["dataset%d/where" % (t + 1)]["elangle"]
    coord[t, ...] = georef.sweep_centroids(nrays, rscale, nbins, elangle)
    # here you get the reflectivity data out of the pvol-object and offset and scale applied
    refl.append(what["offset"] + what["gain"] * pvol["dataset%d/data1/data" % (t + 1)]
# create numpy array (ntilt, nrays, nbins)
refl = np.stack(refl, axis=0)
```
I assume that the reflectivity data is inside `datasetX/data1`. Then `refl` will be a `numpy` array holding your reflectivity data from all elevation/scans inside the volume. 

For better understanding dictionaries and how to inspect them, see https://docs.python.org/3/library/stdtypes.html#mapping-types-dict. Also this gist migth be helpful for data inspection:
http://wradlib.org/2016/03/treeview-with-traits/.

diegocarrio, 2017-07-19:
Thank you very much for your response. I think I have obtained the fields desired. Attached a example of the data I have obtained.
[image]

However, the data contain clutter signal and I would like to remove it. I tryied wradlib.clutter.filter_gabella but I obtained the following error:

In [219]: clutter=wradlib.clutter.filter_gabella(pvol['dataset1/data1/data'],tr1=12,n_p=6,tr2=1.1)
     ...: 
---------------------------------------------------------------------------
OverflowError                             Traceback (most recent call last)
<ipython-input-219-1a48f687fd68> in <module>()
----> 1 clutter=wradlib.clutter.filter_gabella(pvol['dataset1/data1/data'],tr1=12,n_p=6,tr2=1.1)

/opt/anaconda/anaconda2/lib/python2.7/site-packages/wradlib/clutter.pyc in filter_gabella(img, wsize, thrsnorain, tr1, n_p, tr2, rm_nans, radial, cartesian)
    181     if rm_nans:
    182         img = img.copy()
--> 183         img[bad] = np.Inf
    184     ntr1 = filter_gabella_a(img, wsize, tr1, cartesian, radial)
    185     if not rm_nans:

OverflowError: cannot convert float infinity to integer

Could you help me to solve this problem or find a solution to remove false echoes or ground clutter?

Thanks in advance,
Diego Carrió

kmuehlbauer, 2017-08-14:
The data inside `pvol['dataset1/data1/data']` is of type integer and **not** the final reflectivity.

You would need to multiply by `gain` and add the `offset` (see my post before) for every scan:

```python
  refl = pvol["dataset1/data1/what"]["offset"] +  pvol["dataset1/data1/what"]["gain"] * pvol["dataset1/data1/data"]
```

kmuehlbauer, 2017-08-30:
@diegocarrio Did you get along with this? I'll close the issue. Please reopen if appropriate.

closed 2017-08-30 by kmuehlbauer