wradlib/wradlib #168: numpy.stack in read_gdal_values

egouden, 2017-08-01:
Hi all,

This line

return np.squeeze(np.stack(bands))

 requires numpy 1.10.1

This should be equivalent

return np.squeeze(np.array(bands))

Regards,

Edouard

kmuehlbauer, 2017-08-30:
@heistermann How do we proceed here? 

https://github.com/wradlib/wradlib/blob/bea7a2b8fc41dfb6cab3c1edfc15b99b12ee850b/wradlib/georef/raster.py#L208-L220

heistermann, 2017-08-30:
Hm, only thing I can say is that both expressions appear in fact to be equivalent (with numpy.stack having axis=0). As far as I am concerned, we could change the return line to `return np.squeeze(np.array(bands))`. Question is whether we want to generally increase backward compatibility  to older numpy versions - I'd rather say no simply due to the costs.

kmuehlbauer, 2017-08-30:
OK, then I'll add this change in one of the upcoming FIX-PR's. Concerning backwards compatibility I'm we might test on a selection of the most recent (3?) available (conda-forge) numpy versions.

PR #199 (2017-12-13): FIX: use np.array instead of np.stack to be backwards compatible
closes #168
closed 2017-12-13 by kmuehlbauer