wradlib/wradlib #169: version of gdal

jinsanko, 2017-08-10:
Can you advise me the version of GDAL for the newest version of WRADLIB?
I followed the below link to install GDAL 1.11.0 or GDAL 2.2.1.
 https://milkator.wordpress.com/2014/05/06/set-up-gdal-on-ubuntu-14-04/
I got different error message for different version of GDAL..
Below is the error message for GDAL2.2.1.   The libgdal.so.20 is under /usr/local/lib.
Thank. 

Traceback (most recent call last):
  File "typical_workflow.py", line 13, in <module>
    import wradlib
  File "/usr/local/lib/python2.7/dist-packages/wradlib/__init__.py", line 31, in <module>
    from . import adjust  # noqa
  File "/usr/local/lib/python2.7/dist-packages/wradlib/adjust.py", line 112, in <module>
    import wradlib.ipol as ipol
  File "/usr/local/lib/python2.7/dist-packages/wradlib/ipol.py", line 45, in <module>
    from . import util as util
  File "/usr/local/lib/python2.7/dist-packages/wradlib/util.py", line 38, in <module>
    from osgeo import ogr
  File "/usr/local/lib/python2.7/dist-packages/osgeo/__init__.py", line 21, in <module>
    _gdal = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/osgeo/__init__.py", line 17, in swig_import_helper
    _mod = imp.load_module('_gdal', fp, pathname, description)
ImportError: libgdal.so.20: cannot open shared object file: No such file or directory

kmuehlbauer, 2017-08-14:
@jinsanko First, how did you install `wradlib`?

`wradlib` should work with GDAL version > 1.9, but we recommend > 2.1.0 (for performance). The problem you face, is that your `python` doesn't pick up/find the correct `libgdal.so.20`. This might be a library search PATH issue of your system, but I really have no idea.

We usually recommend using `conda` (http://wradlib.org/wradlib-docs/latest/gettingstarted.html#installation) to get rid of specific system's issues.

kmuehlbauer, 2017-08-14:
Closing, as not a wradlib but GDAL issue.

closed 2017-08-14 by kmuehlbauer

jinsanko, 2017-08-14:
Thank. I will retry it again.