wradlib/wradlib #171: Inconsistent shapes of arrays returned by different classes in the ipol module.

heistermann, 2017-08-25:
In case only one set of source values is passed to the `__call__` method, different classes in `wradlib.ipol` behave differently where they should behave consistently.

`Nearest` and `Idw` return 1-D (flat) arrays, `OrdinaryKriging` and `ExternalDriftKriging` return matrices of shape (numtargetpoints, 1). This causes to problems e.g. when `OrdinaryKriging` is used as argument `Ipclass` in the `adjust` modules adjustment classes (`MemoryError` when matrix is multiplied with a flat array).

We might have not explicitely caught this mahevious so far because unit tests for `Idw` and `Nearest` are not implemented, and `OrdinaryKriging` has not been routinely used as an interpolation class when using classes from `adjust`.

Fixing this is ambivalent: The idea has always been that you can pass multiple source datasets in one 2-D array (num sourcepoints, num datasets) to the `__call__` methods. 

*Option 1:* I'd prefer to change `Idw` and `Nearest` so they return an array of shape (mum sourcepoints, 1) in case only one dataset is passed. This might, however, break several applications who have been using these classes so far. 

*Option 2:* Changing `OrdinaryKriging` to behave like e.g. `Idw` might break applications using `OrdinaryKriging`.

*Option 3*: The number of dimensions of the returned array should be the same as the number of dimensions of the input array. This might still break some applications: if people have so far been passing flat arrays to `Idw`, they will still get back flat arrays - so no problem. But if they have been passing a flat array to `OrdinaryKriging`, they will get back a flat array - which might still break their applications. 

My suggestion: Use Option 3, but not as a bug fix release, but with a new 1.0 release. What do you think, @kmuehlbauer ?

kmuehlbauer, 2017-08-25:
@heistermann If there is the slightest chance that things brake after the change, then we should take Option 3.

kmuehlbauer, 2017-08-25:
@heistermann Would also be a good opportunity to implement tests.

heistermann, 2017-08-25:
Yes, definetely. As pointed out, option 3 might also break applications, but we can explicetely state this with a new release. It is also more intuitive for those who have been using this with flat arrays. It would only require to ravel the out put from the Kriging classes in case only one dataset is used.

I will make the documentation more explicit in that regard, and, yes, implement tests for the other classes.

kmuehlbauer, 2018-02-24:
resolved in #226

closed 2018-02-24 by kmuehlbauer