wradlib/wradlib #182: Import error Anaconda 3.6.2

GISSruhl, 2017-10-07:
Was not able to import wradlib after following install instructions. 

G:\Anaconda\Scripts>activate wradlib

(wradlib) G:\Anaconda\Scripts> conda install wradlib
Fetching package metadata ...............
Solving package specifications: .

# All requested packages already installed.
# packages in environment at G:\Anaconda\envs\wradlib:
#
wradlib                   0.10.1                   py36_0    conda-forge

(wradlib) G:\Anaconda\Scripts> python
Python 3.6.2 |Anaconda custom (64-bit)| (default, Sep 19 2017, 08:03:39) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import wradlib
G:\Anaconda\envs\wradlib\lib\site-packages\osgeo\__init__.py:12: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "G:\Anaconda\envs\wradlib\lib\site-packages\wradlib\__init__.py", line 31, in <module>
    from . import adjust  # noqa
  File "G:\Anaconda\envs\wradlib\lib\site-packages\wradlib\adjust.py", line 112, in <module>
    import wradlib.ipol as ipol
  File "G:\Anaconda\envs\wradlib\lib\site-packages\wradlib\ipol.py", line 45, in <module>
    from . import util as util
  File "G:\Anaconda\envs\wradlib\lib\site-packages\wradlib\util.py", line 38, in <module>
    from osgeo import ogr
  File "G:\Anaconda\envs\wradlib\lib\site-packages\osgeo\__init__.py", line 25, in <module>
    _gdal = swig_import_helper()
  File "G:\Anaconda\envs\wradlib\lib\site-packages\osgeo\__init__.py", line 21, in swig_import_helper
    _mod = imp.load_module('_gdal', fp, pathname, description)
  File "G:\Anaconda\envs\wradlib\lib\imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "G:\Anaconda\envs\wradlib\lib\imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: DLL load failed: The specified module could not be found.

kmuehlbauer, 2017-10-08:
Please post the output of `conda list` within your `wradlib` environment. This is most likely a channel problem.

GISSruhl, 2017-10-08:
G:\Anaconda\Scripts>activate wradlib

(wradlib) G:\Anaconda\Scripts> conda list
# packages in environment at G:\Anaconda\envs\wradlib:
#
_ipyw_jlab_nb_ext_conf    0.1.0            py36he6757f0_0
alabaster                 0.7.10           py36hcd07829_0
anaconda                  custom                   py36_0
anaconda-client           1.6.5            py36hd36550c_0
anaconda-navigator        1.6.8            py36h4b7dd57_0
anaconda-project          0.8.0            py36h8b3bf89_0
asn1crypto                0.22.0           py36h8e79faa_1
astroid                   1.5.3            py36h9d85297_0
astropy                   2.0.2            py36h06391c4_4
babel                     2.5.0            py36h35444c1_0
backports                 1.0              py36h81696a8_1
backports.shutil_get_terminal_size 1.0.0            py36h79ab834_2
beautifulsoup4            4.6.0            py36hd4cc5e8_1
bitarray                  0.8.1            py36h6af124b_0
bkcharts                  0.2              py36h7e685f7_0
blaze                     0.11.3           py36h8a29ca5_0
bleach                    2.0.0            py36h0a7e3d6_0
bokeh                     0.12.7           py36h012f572_1
boto                      2.48.0           py36h1a776d2_1
bottleneck                1.2.1            py36hd119dfa_0
bzip2                     1.0.6            vc14hdec8e7a_1  [vc14]
ca-certificates           2017.08.26           h94faf87_0
cachecontrol              0.12.3           py36hfe50d7b_0
certifi                   2017.7.27.1      py36h043bc9e_0
cffi                      1.10.0           py36hae3d1b5_1
chardet                   3.0.4            py36h420ce6e_1
click                     6.7              py36hec8c647_0
cloudpickle               0.4.0            py36h639d8dc_0
clyent                    1.2.2            py36hb10d595_1
colorama                  0.3.9            py36h029ae33_0
comtypes                  1.1.2            py36heb9b3d1_0
conda-verify              2.0.0            py36h065de53_0
console_shortcut          0.1.1                haa4cab3_2
contextlib2               0.5.5            py36he5d52c0_0
cryptography              2.0.3            py36h123decb_1
curl                      7.55.1           vc14hdaba4a4_3  [vc14]
cycler                    0.10.0           py36h009560c_0
cython                    0.26.1           py36h18049ac_0
cytoolz                   0.8.2            py36h547e66e_0
dask                      0.15.2           py36h25e1b01_0
dask-core                 0.15.2           py36hf9e56b0_0
datashape                 0.5.4            py36h5770b85_0
decorator                 4.1.2            py36he63a57b_0
distlib                   0.2.5            py36h51371be_0
distributed               1.18.3           py36h5be4c3e_0
docutils                  0.14             py36h6012d8f_0
entrypoints               0.2.3            py36hfd66bb0_2
et_xmlfile                1.0.1            py36h3d2d736_0
expat                     2.2.1                    vc14_0  [vc14]  conda-forge
fastcache                 1.0.2            py36hffdae1b_0
filelock                  2.0.12           py36hd7ddd41_0
flask                     0.12.2           py36h98b5e8f_0
flask-cors                3.0.3            py36h8a3855d_0
freetype                  2.8              vc14h17c9bdf_0  [vc14]
freexl                    1.0.2                    vc14_2  [vc14]  conda-forge
gdal                      2.2.0           np111py36_vc14_2  [vc14]  conda-forge
geos                      3.5.1                    vc14_1  [vc14]  conda-forge
get_terminal_size         1.0.0                h38e98db_0
gevent                    1.2.2            py36h342a76c_0
glob2                     0.5              py36h11cc1bd_1
greenlet                  0.4.12           py36ha00ad21_0
h5py                      2.7.1                    py36_1    conda-forge
hdf4                      4.2.12                   vc14_0  [vc14]  conda-forge
hdf5                      1.8.18                   vc14_0  [vc14]  conda-forge
heapdict                  1.0.0            py36h21fa5f4_0
[... 139 lines cut ...]
tornado                   4.5.2            py36h57f6048_0
traitlets                 4.3.2            py36h096827d_0
typing                    3.6.2            py36hb035bda_0
unicodecsv                0.14.1           py36h6450c06_0
urllib3                   1.22             py36h276f60a_0
vc                        14                   h2379b0c_1
vs2015_runtime            14.0.25123           hd4c4e62_1
wcwidth                   0.1.7            py36h3d5aa90_0
webencodings              0.5.1            py36h67c50ae_1
werkzeug                  0.12.2           py36h866a736_0
wheel                     0.29.0           py36h6ce6cde_1
widgetsnbextension        3.0.2            py36h364476f_1
win_inet_pton             1.0.1            py36he67d7fd_1
win_unicode_console       0.5              py36hcdbd4b5_0
wincertstore              0.2              py36h7fe50ca_0
wradlib                   0.10.1                   py36_0    conda-forge
wrapt                     1.10.11          py36he5f5981_0
xerces-c                  3.1.4                    vc14_2  [vc14]  conda-forge
xlrd                      1.1.0            py36h1cb58dc_1
xlsxwriter                0.9.8            py36hdf8fb07_0
xlwings                   0.11.4           py36hd3cf94d_0
xlwt                      1.3.0            py36h1a4751e_0
yaml                      0.1.7            vc14hb31d195_1  [vc14]
zict                      0.1.2            py36hbda90c0_0
zlib                      1.2.11           vc14h1cdd9ab_1  [vc14]

kmuehlbauer, 2017-10-09:
@GISSruhl You have data from `defaults` and `conda-forge` channel inside your environment. This is known to create errors like you are facing.

Did you clone the root environment (as written in the docs)? @heistermann This might lead to channel clashing problems, if the root environment is from `defaults`. We might need changing the docs.

Be sure to use a recent `conda`. Please check your [.condarc](https://conda.io/docs/user-guide/configuration/use-condarc.html) if `conda-forge` is on top of your channels (highest priority). Then re-create your environment from scratch.

```bash
> conda update conda
> conda config --get channels
> conda create -n wradlib python=3.6
> activate wradlib
> conda install wradlib
```
Please report back, if this doesn't help.

heistermann, 2017-10-09:
I agree - I think there is no substantial advantage in cloning the root environment when creating a wradlib environment.

GISSruhl, 2017-10-10:
That seemed to help, I'm no longer failing on import.  I had to uninstall wradlib, then delete the folder out of Anaconda\env before re-installing.  It now shows as packaged by conda-forge.  

I am getting some deprecation warnings on import. Let me know if I should keep it under this issue or open a new one.

(wradlib) G:\Anaconda\Scripts>python
Python 3.6.3 | packaged by conda-forge | (default, Oct  5 2017, 14:39:21) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import wradlib
G:\Anaconda\envs\wradlib\lib\site-packages\scipy\_lib\_numpy_compat.py:287: DeprecationWarning: invalid escape sequence \p
  """
G:\Anaconda\envs\wradlib\lib\site-packages\osgeo\__init__.py:12: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
G:\Anaconda\envs\wradlib\lib\site-packages\scipy\signal\filter_design.py:425: DeprecationWarning: invalid escape sequence \p
  """
>>>

Thank you so much for your quick help!

kmuehlbauer, 2017-10-10:
@GISSruhl Glad, that it worked for you. We'll change the misleading docs-section.

The DeprecationWarnings are not wradlib related. They are originating from dependencies (scipy, gdal). We can't do much about that.

closed 2017-10-10 by kmuehlbauer