wradlib/wradlib #190: Considering interpolation effects on Beamblockage Calculation using DEM

kmuehlbauer, 2017-11-30:
For calculation of beamblockage wradlib interpolates the cartesian DEM grid cells to the polar radar raster via a b-spline approach. Due to this the DEM grid cells are averaged to one single value and might not give the precise beamblockage fraction in the further calculation after [Bech et al.](http://journals.ametsoc.org/doi/pdf/10.1175/1520-0426%282003%29020%3C0845%3ATSOSPW%3E2.0.CO%3B2)

We should explore how this effects the overall performance in terms of introduced error. We could try to implement a kind of DEM azimuthal cut for every range and calculate the beamblockage via this approach.

Also we should think about accessing the faktor k_e which contains the atmospheric propagation conditions in order to calculate beamblockage for different refractive index.

Thanks @tsmsalper for the offline discussion.

heistermann, 2017-12-11:
First step would be to transform the 3D DEM coordinates into polar coordinates "as seen" from the radar, right? Like an analogon to [sat2pol](http://wradlib.org/wradlib-docs/latest/generated/wradlib.georef.satellite.sat2pol.html#wradlib.georef.satellite.sat2pol), or an "inverse" to [polar2lonlatalt_n](polar2lonlatalt_n). Then we could project the DEM along the beam basically by collapsing the range coordinates, right?  Does that sound confusing/confused?

kmuehlbauer, 2017-12-11:
Makes sense. This way we could get the indices of all DEM cells withing the beam (for all ranges) just by subscripting the wanted angle. 

But it is not clear to me, how you want to "collapse" the range coordinates.

kmuehlbauer, 2017-12-11:
OK, think I know what you mean. If we have all those DEM-Cells within one beam (azimuth, range, elevation), we "collapse" the range the range coordinates and get an array of (azimuth, elevation) values. With these values we can retrieve a vertical-cut-projection of the beam blockage.

Nice approach!

kmuehlbauer, 2018-11-01:
This is implemented as example in [wradlib/beamblockage](https://github.com/wradlib/beamblockage)

This would need to be generalized and added to wradlib codebase.

gewitterblitz, 2021-03-04:
I could not reproduce the plots in beam blockage tutorial available at this [link](https://docs.wradlib.org/en/1.0.0/notebooks/beamblockage/wradlib_beamblock.html#Read-DEM-Raster-Data).

Rather, running the same notebook using `geo/bonn_new.tif` file gave me this output.

[image]

kmuehlbauer, 2023-02-13:
I might come back to this in wradlib 2.0, with an xarray-only approach.

idirdehmous, 2024-09-11:
Dear Kmuelbauer ,
I have already used wradlib 3 years ago ,, now i am using it again to read and process the ODIM belgian Radars .
My objective to plot the Beam blockage fraction. So i am following this example :

https://docs.wradlib.org/en/latest/notebooks/beamblockage/beamblockage.html

However i got stuck at this line . 
# Plot terrain (on ax1)
polarvalues = wrl.georef.create_xarray_dataarray(
    polarvalues, r=r, phi=coord[:, 0, 1]
).wrl.georef.georeference()
dem = polarvalues.wrl.vis.plot(ax=ax1, cmap=mpl.cm.terrain, vmin=0.0)

The function   ipol_to_irrigular_spline returns strange polaralues 
In addition , in some examples i found wrl.vis.plot(......) and others , wrl.vis.plot_ppi (......) 

The questions are : 
Did you already find a bug in this function ? 
Which version of wradlib did you use to plot the raster polar values and the CBB ? 

Thank you 
Idir

idirdehmous, 2024-09-12:
@kmuehlbauer , sorry .
I will open a new one .
Cheers