wradlib/wradlib #219: logging config

egouden, 2018-02-20:
Hi,

At some point my logs were gone and it took me some time to find the origin of the problem ;-).
The user, not the library, should dictate what happens when a logging event occurs :

http://docs.python-guide.org/en/latest/writing/logging/

Therefore logging.basicConfig() should not be used in wradlib.
Could you please remove commit d8df90ab67bf2f06edbcc8345436fd5f5e68eff3 ?

Thank you in advance,

Edouard

heistermann, 2018-02-20:
Hey Edouard,

good to hear from you. I assume that you would basically require to remove line 32 from the [atten module](https://github.com/wradlib/wradlib/blob/master/wradlib/atten.py) (i.e. `logging.basicConfig()`) But I guess keeping line 33 

```
logger = logging.getLogger('attcorr')
``` 
should be fine? Or, according to docs, rather

```
logger = logging.getLogger(__name__)
``` 

From a user perspective, would you also require a logger in the `__init__.py` module, or in other modules?

To be honest, I have not though about systematically adding logging capabilities to wradlib. 

@kmuehlbauer : Would that be another issue for v1.0?

kmuehlbauer, 2018-02-20:
Hi @egouden, glad you are an ongoing wradlib user.

@heistermann This is a definite issue for 1.0

egouden, 2018-02-20:
Hi Maik

Thank you for the quick reply!
Yes, I would remove line 32.
I don't have specific logging needs and don't think it is really mandatory.

Regards,

Edouard

kmuehlbauer, 2018-02-20:
@heistermann I can do this, have to change some things tomorrow...

kmuehlbauer, 2018-02-21:
@egouden This is fixed in latest github and also in https://github.com/wradlib/wradlib/releases/tag/v0.11.3.

closed 2018-02-21 by kmuehlbauer

egouden, 2018-02-22:
This was quick !

Thank you.