wradlib/wradlib #228: Manually installing requirements - irritating installation proceedure

s-m-e, 2018-02-26:
Is there any particular reason why requirements as specified in `requirements.txt` and `requirements_devel.txt` are not installed when `setup.py` is executed? This is quite irritating. If one installs `wradlib` with `pip` for instance, one must *manually* install all dependencies listed in `requirements.txt` beforehand. 

Patching `setup.py` along the following lines would remove the irritations:
```
with open('requirements.txt', 'r') as f:
	runtime_requirements = [rq for rq in f.read().split('\n') if rq != '']
with open('requirements_devel.txt', 'r') as f:
	development_requirements = [rq for rq in f.read().split('\n') if rq != '']
setup(
	# [...]
	install_requires = runtime_requirements,
	extras_require = {'dev': development_requirements},
	# [...]
	)
```

kmuehlbauer, 2018-02-26:
@s-m-e There is a particular reason. We have several dependencies, namely h5py, netcdf4 and gdal which rely on their c/c++ libraries. Those can't be installed via pip. Therefore we suggest to install all dependencies beforehand or use conda-forge.

s-m-e, 2018-02-26:
I understand your point. However, it makes matters more complicated when I add `wradlib` as a dependency to another project. I can not just make it a regular dependency - instead, I must either "hack" the installation procedure of my project in some non-standard fashion or add `wradlib`'s dependencies as direct dependencies to my project (as opposed to implicit, indirect dependencies). Both ways feel wrong ... What's your recommended approach?

(I have seen other projects similar to `wradlib` which explicitly state that they rely on e.g. `h5py` - therefore `libhdf5` must be installed. They subsequently provide simple guidelines on how to install the required library with package managers of some widely used Linux or BSD distributions, for instance `apt-get install libhdf5-dev libhdf5 python-dev python3-dev` for Debian/Ubuntu.)

kmuehlbauer, 2018-02-26:
@s-m-e I get your point too. Seems, we need to fix this on our side.

@heistermann As we are using setup.py, we need to apply to the rules. WDYT?

heistermann, 2018-02-27:
Hm, not sure... apply the rules, yes, but what *are* the rules? In my point of view, we made a deliberate decision to not support installation via `pip` anymore and therefore remove any kind of dependency management from `setup.py`. Specifying a dependency such as `h5py` in `setup.py` and letting the user take care of additional libraries was causing the trouble.

Now, my question is: Can we actually support `pip` without breaking our workflow with `conda-forge`? In other words: If we followed the recommendation of @s-m-e to specify e.g. `h5py` in `setup`'s `install_requires`, what would happen if `python setup.py install` were executed *after* dependencies have been taken care of via `conda-forge`? Could it break that installation, or will `setup` recognize that dependencies are already met and just leave it be?

Sorry if that all sounds awkward... I am trying to put this in my own words.

kmuehlbauer, 2018-02-27:
@heistermann The more I read the more confusing this gets.

- `numpy.distutils.core.setup` does not have `install_requires`.

kmuehlbauer, 2018-02-27:
@heistermann @s-m-e 

OK, a little bit of testing and there might be a working solution.

Although, `numpy.distutils.core.setup` does not have `install_requires`, pip will handle this correctly.

I added the `install_requires` key with the dependencies.

Case 1 - conda-env with all dependencies installed:
    - `python setup.py install` - same as before

Case 2 - conda-env with all dependencies installed:
    - `python setup.py sdist` - build sdist
    - `pip install dist/wradlib-1.0.0.tar.gz`- works since pip uses the correct `setup` under the hood.

Case 3 - conda-env with only `gdal` installed (includes numpy, hdf5 and netcdf)
     -  - `pip install dist/wradlib-1.0.0.tar.gz`- missing parts are downloaded as wheels and correctly processed.

So, it seems, that we just need to add correct `install requires`.

heistermann, 2018-02-27:
Which could be stripped from `requirements.txt` as suggested by @s-m-e ?

kmuehlbauer, 2018-02-27:
@heistermann Yes, exactly. One thing, though. If you install via `python setup.py install` in an **empty** environment, then it complains about missing numpy. We can add a try/except there and use `from distutils.core import setup` when numpy isn't available.

kmuehlbauer, 2018-03-03:
fixed by #229

PR #229 (2018-02-28): Wip docs 2
- remove not needed template, css and docs
- fix notebook
- fix #228
closed 2018-03-03 by kmuehlbauer