wradlib/wradlib #240: Default arguments in correctAttenuationHB are buggy

heistermann, 2018-03-13:
The current implementation of `wradlib.atten.correctAttenuationHB` looks like

```
def correctAttenuationHB(gateset,
                         coefficients={'a': 1.67e-4, 'b': 0.7, 'l_rg': 1.0},
                         mode='except',
                         thrs=59.0)
    ...
    gate_length = coefficients['gate_length']
   ...
```

I suggest to keep the function body as is, but to correct the default argument like 

```
def correctAttenuationHB(gateset,
                         coefficients={'a': 1.67e-4, 'b': 0.7, 'gate_length': 1.0},
                         mode='except',
                         thrs=59.0)
    ...
    gate_length = coefficients['gate_length']
   ...
``` 

which would also be consistent with the other functions in atten.

The bug will only show if the function is called with default arguments...

kmuehlbauer, 2018-03-13:
@heistermann Is this for V1.0.0 ? Then we can change it while revising the `atten`-module.

heistermann, 2018-03-13:
As far as I am concerned, we assign this to 1.0.
But then again, it's a real bug, so I guess we should also fix it for a final 0.11.x release.

kmuehlbauer, 2018-03-28:
resolved by #245

closed 2018-03-28 by kmuehlbauer