wradlib/wradlib #264: "rstart" in Recipe #2 should be multiplied with 1000

overeem11, 2018-04-05:
Dear colleague,

At KNMI we noticed that in Recipe #2 ("Reading and visualizing an ODIM_H5 polar volume"; http://docs.wradlib.org/en/latest/notebooks/workflow/recipe2.html) "rstart" should be multiplied with 1000. According to the OPERA report "EUMETNET OPERA weather radar information model for
implementation with the HDF5 file format" (http://eumetnet.eu/wp-content/uploads/2017/01/OPERA_hdf_description_2014.pdf) the unit of "rstart" is km, whereas the unit of "rscale" is m. Hence, we believe that in the following lines of the recipe "rstart" should be multiplied with 1000:
  r = np.arange(where["rstart"],
                  where["rstart"] + where["nbins"] * where["rscale"],
                  where["rscale"])

If the value of "rstart" is 0, as is e.g. the case for the KNMI radar data, this missing conversion factor is, of course, of no consequence.

Best regards,
Aart Overeem
KNMI

kmuehlbauer, 2018-04-05:
@overeem11 Thanks Aart for pointing this out. And thanks for the pointer to the OPERA description. We'll have this changed soon in https://github.com/wradlib/wradlib-notebooks.

closed 2018-04-06 by kmuehlbauer

kmuehlbauer, 2018-04-06:
@overeem11 This is now resolved in latest https://github.com/wradlib/wradlib-notebooks. 
Please see also http://docs.wradlib.org/en/latest/notebooks/workflow/recipe2.html.

Thanks for helping in making wradlib better :+1: