wradlib/wradlib #267: read_Rainbow not closing file properly

meteoswiss-mdr, 2018-04-10:
Dear wradlib masters,

When using wradlib.io.read_Rainbow to read a rainbow file I get the following warning:
unclosed file <_io.BufferedReader name='[name_of_file.vol]'>

I am using wradlib version 0.11.3 with python version 3.6.4.

Could you please have a look at it?

Thanks!

kmuehlbauer, 2018-04-10:
@meteoswiss-mdr We would need the rainbow source file to debug this.

meteoswiss-mdr, 2018-04-10:
@kmuehlbauer, can I send it by private email?

kmuehlbauer, 2018-04-10:
@meteoswiss-mdr Sure. I think you have my email address. Otherwise, you'll find it via the link in my github-bio.

kmuehlbauer, 2018-04-10:
@heistermann If `read_rainbow` is called with a string, the file is opened, but never closed. This might lead to this **Warning**. There are also other places where we do not properly close files, we have opened. We should adress this for the next minor release. WDYT?

heistermann, 2018-04-10:
Absolutely. Maybe we can switch, at least for some cases, to use the `with` statement as it also handles closing in case of Exception that occur on the way before the `close` statement. Like
```
with open("x.txt") as f:
    data = f.read()
    do something with data
```

kmuehlbauer, 2018-04-10:
Yes, I checked with @meteoswiss-mdr that this would be the most pythonic way. But we would need to see how this fits when we call with filehandle.

kmuehlbauer, 2018-04-10:
@heistermann, We already use this construct in the tests.

kmuehlbauer, 2018-07-19:
@meteoswiss-mdr Finally looking into this, I'm trying to recreate the warning with no luck. Could you provide a minimal example showing which issues this warning?

PR #275 (2018-08-02): FIX: #267
- use with-statement in `wradlib.io.rainbow.read_rainbow()`
- remove accidentally merged `logs`-file in `wradlib.io`

closes #267
commit 4361cd2e (2018-08-02): FIX: #267 (#275)

* FIX: use with-statement in `wradlib.io.rainbow.read_rainbow()`, remove logs-file

* FIX: move conditional `loaddata` into with-statement, fix dual use of f
closed 2018-08-02 by kmuehlbauer