wradlib/wradlib #272: Implement ODIM Reader/Writer

kmuehlbauer, 2018-07-06:
Implement ODIM Reader/Writer from `rmilib` (suggested by @egouden)

kmuehlbauer, 2018-07-06:
@egouden Thank you again for the quick introduction on your current work in `rmilib` at ERAD2018. We can track the upcoming implementation work in this issue.

kmuehlbauer, 2018-11-01:
@egouden Is there anything we can do to get things along? We could create a private repo here for the implementation process. Would that be a way to go for you?

egouden, 2018-11-02:
Hello,

Good to hear you!

Unfortunately the only blocking factor is time. I am mainly busy with
support and maintenance of my precipitation products, preparing
presentations and coordinating projects. I have hardly time for big
development. In fact the Odim module is my top priority development.
I want to finnish this very soon.
I already checked the Odim validator.
I give you the code next week to start collaborating in public 😉

See you soon.

Edouard

kmuehlbauer, 2019-02-15:
resolved by #311

closed 2019-02-15 by kmuehlbauer