wradlib/wradlib #286: Any possibility to speedup the ipol.Idw(vpr.CAPPI) module?

sysumeteo, 2018-09-25:
The vpr.CAPPI module took me more than 2 minutes to interpolating one field of a radar into a 13*921*921 grid.It is quite a long time for large amount data processing.Is there any possibility to speedup the ipol.Idw procedure? For example,just like the speedup.f  for PhiDP unfolding.

heistermann, 2018-09-25:
Yes, the performance is quite poor, but as of now, there is no way to speed this up out of the box. Sorry. Under the hood, the interpolation relies on a neighbourhood search carried out by `scipy.spatial.cKDTRee`. I guess there is potential for optimisation in our code even without moving this into a Fortran implementation. But right now, this is all we have...

sysumeteo, 2018-09-25:
> Yes, the performance is quite poor, but as of now, there is no way to speed this up out of the box. Sorry. Under the hood, the interpolation relies on a neighbourhood search carried out by [...]

Thanks for your prompt reply.Operationaly,maybe we can carry out the interpolation at different z-levels vias multithread processing,and then merge the output  together to speed up the procedure,if memory and CPU resource are enough.But unfortunately i notice that the minimal altitude of vpr.CAPPI interpolation is fixed  to  0.  I think this may make trouble to the one who just want  to interpolate data to the upper part of the grid.

heistermann, 2018-09-25:
Not sure if I understand - could you please elaborate further? I have to admit I did not use the `vpr` module in quite a long while. Doesn't `vpr.CAPPI` just take the arrays of source and target points? Can't you specify a target grid at any altitude in order to split up the job into different altitudes?

winash12, 2018-09-25:
If you look at the source code of https://github.com/scipy/scipy/blob/master/scipy/spatial/ckdtree.pyx

it is already doing multi thread processing under the hood. I am not sure which API you are using but query_ball_point is implemented in C++ and the threading is being done.

sysumeteo, 2018-09-25:
> Not sure if I understand - could you please elaborate further? I have to admit I did not use the `vpr` module in quite a long while. Doesn't `vpr.CAPPI` just take the arrays of source and target [...]

That maybe my mis-understanding!
What i mean above is about the module 'vpr.make_3D_grid' which only accept 'maxalt' as input,and then output a grid  that start  at 0 in z-levels. But  anyway , it doesn't matter.I can slice it  or construct  grid  by myself as the input of  vpr.CAPPI.

heistermann, 2018-09-25:
Oh, I see. That we can fix. I have created an issue #287 accordingly. Please watch for corresponding PRs.

kmuehlbauer, 2018-09-25:
If we look into the code, the long operation time is mostly due to the creation and query of the `cKDTree`.
We could add some convenience to make it easier to re-use the interpolation/gridder object (e.g. in CAPPI). In most cases the grid-layout is the same. Do I miss something?

heistermann, 2018-09-26:
@kmuehlbauer : I think this behaviour is already implemented. The `cKDTree` is created as part of constructing an interpolator in the `__init__` method of the class (e.g. `CAPPI`), i.e. when the volume object is defined. After that you just `__call__` that object repeatedly with new polar data - without reconstructing the interpolator object / the cKDTree. That behaviour is dominated by `wradlib.ipol`. The volume classes are basically wrappers around that, including some masking depending on what you want to have in your finaly Cartesian volume.

kmuehlbauer, 2018-09-26:
@heistermann True, but you have to create the interpolator instance again and again for different CAPPI's from the same dataset. There I see space for improvement.

heistermann, 2018-09-26:
Hm, not sure if I understand...again. The interpolator/CAPPI instance is created for a defined set of source (polar) and target (Cartesian 3D grid) coordinates. That is done once. If you want to parallelize, you can split the 3D grid into different vertical slices of e.g. shape (1, ny, nx) to create CAPPI/interpolator instances. But I don't really see how these can benefit from each other. But maybe **I** am missing something?

heistermann, 2018-09-27:
Ok, now I understand. Although I am not entirely sure how much computational resources go into (1) creating the `cKDTree` and (2) querying the `cKDTree`, it should definitely speed up initiation of multiple CAPPI objects if we did `cKDTree(src)` only once. Would you suggest to add another keyword  argument for the `__init__` methods of the interpolation classes? Like e.g. for `Nearest`

```
    def __init__(self, src, trg, kdtree=None):
        src = self._make_coord_arrays(src)
        trg = self._make_coord_arrays(trg)
        ...
        # plant a tree
        if kdtree is None:
            self.tree = cKDTree(src)
        else:
            self.tree = kdtree
        self.dists, self.ix = self.tree.query(trg, k=1)
```

kmuehlbauer, 2018-09-27:
@heistermann [This example](https://docs.wradlib.org/en/stable/notebooks/workflow/recipe2.html):

tree-creation
```
CPU times: user 1min 10s, sys: 7.35 ms, total: 1min 10s
Wall time: 1min 11s
```

tree-query:
```
CPU times: user 1.05 s, sys: 1.87 ms, total: 1.06 s
Wall time: 1.05 s
```

kmuehlbauer, 2018-09-27:
@heistermann OMG!

kmuehlbauer, 2018-09-27:
Maybe we do not need this stuff at all.

If we create the cKDTree like this:

```python
tree = cKDTree(src, balanced_tree=False)
```
the creation time drops to:
```
CPU times: user 218 ms, sys: 14 µs, total: 218 ms
Wall time: 216 ms
```

kmuehlbauer, 2018-09-27:
The query-time is only minimal higher (1.23 seconds all)

heistermann, 2018-09-27:
WHAT?? Unbelievable.

kmuehlbauer, 2018-09-28:
@sysumeteo We have a working prototype which is really speeding up things quite substantially. I hope to get it into the code base early next week.

sysumeteo, 2018-09-28:
> @sysumeteo We have a working prototype which is really speeding up things quite substantially. I hope to get it into the code base early next week.

This is really a good news. Now i can see the ray of hope to deal with my large dataset within the limited time. @kmuehlbauer

kmuehlbauer, 2018-09-28:
@sysumeteo If you want to have a look -> #289.

kmuehlbauer, 2018-10-02:
resolved by #289

closed 2018-10-02 by kmuehlbauer

kmuehlbauer, 2018-10-02:
@sysumeteo You might use the current wradlib github-master for a significant speedup.

See this [docs here](https://docs.wradlib.org/en/stable/notebooks/workflow/recipe2.html
) before the change and [this docs](https://docs.wradlib.org/en/latest/notebooks/workflow/recipe2.html) after the speedup.

`3-D interpolation took: 0:01:53.725399` vs. `3-D interpolation took: 0:00:01.094116`