wradlib/wradlib #293: Add a px250 dwd radar data parser.

meteoDaniel, 2018-10-08:
DWD offers high resolution radar data from all german sites. It is encoded in bufr and with DWD bufr tables and eccodes it is possible to decode it. 

I am having no idea how to interpret the data. 

Here you can find the data 
https://opendata.dwd.de/weather/radar/sites/px250/

meteoDaniel, 2018-10-09:
First you have to install `eccodes` and download the bufr tables from german weather service [https://www.dwd.de/DE/leistungen/opendata/hilfe.html](url). 

Then you have to re-structure the paths of the tables as mentioned in the attached jira issue from th ecmwf. 
[SUP-2141.pdf](https://github.com/wradlib/wradlib/files/2459064/SUP-2141.pdf)

Then you have to set the env. variable to the directory with the bufr tables
`export ECCODES_DEFINITION_PATH=$(pwd):~/utils/eccodes-2.8.0-Source/definitions`

```
import json 
import subprocess
import pandas as pd

bufr_data = subprocess.run(
            f"bufr_dump -jf {file_name}",
            stdout=subprocess.PIPE,
            check=True,
            shell=True
        ).stdout

json_data = json.loads(bufr_data)

radar_data = pd.DataFrame(json_data['messages'])
radar_data.to_pickle('radar_data.p')
```
I attached all used data.  
[radar_data.zip](https://github.com/wradlib/wradlib/files/2459423/radar_data.zip)

Maybe you could check the odim program to transform the data into hdf. I am not able to use the software yet.

kmuehlbauer, 2018-10-09:
@meteoDaniel First, this is extremely annoying that DWD isn't able to provide correct bufr tables. :disappointed: Second, can you also provide the dumped file? This would be easier for me to check.

meteoDaniel, 2018-10-09:
The dumped data is directly parsed into the python object `bufr_data`.
With the code you should be able to reproduce the pickle I attached,

kmuehlbauer, 2018-10-09:
Obviously only metadata is decoded. There are some undecoded descriptors showing up in the bufr dump.
There is nothing we can do about that.

DWD need to fix their bufr tables. This is tremendously annoying. Personally I gave up on that issue.

closed 2018-10-09 by kmuehlbauer