wradlib/wradlib #301: Reading data from IRIS ingest files

aschueth, 2018-11-02:
I would like to be able to read data from binary IRIS ingest files, that are not in the same format as the product_raw files, etc... The format for them is a main header file for that sweep, and then a different file for every radar moment. A sample dataset can be found here: https://drive.google.com/drive/folders/18w3QBu246BoNvPvCTGN4cCaz1I5XwJZR?usp=sharing

While the data linked are PPIs, I also have RHIs that I would like to read in as well.

kmuehlbauer, 2018-11-03:
@aschueth Thanks for opening this issue and providing the ingest data files.

For implementation we will use numpy memmap to get hold of the files contents (as it is done already for the normal sigmet/IRIS files).

Unfortunately we can't use the record approach since the ingest files doesn't have record structure. But we can use all the implemented decoding mechanism.

From my perspective it would be sufficient to create reader functions for the ingest header file and for the ingest data files. So we do not need to put that much magic inside.

I'll link the implementation PR here early next week.

@aschueth Can I contact you offlist regarding a working notebook prototype?

aschueth, 2018-11-03:
Absolutely

aschueth, 2018-11-03:
Absolutely! My email can be found at my account homepage

kmuehlbauer, 2018-11-05:
@aschueth I just sent you an email with a working prototype, so you can start reading the ingest data and metadata.

Since the ingest data has no record structure reading is a bit different from the other sigmet files. It might be necessary to re-think the class structure a bit. Just notifying you, @heistermann, that I might contact you regarding this.

kmuehlbauer, 2018-11-09:
@aschueth If you find the time you might test against my [feature branch](https://github.com/kmuehlbauer/wradlib/tree/sigmet-redesign). See PR #303.

aschueth, 2018-11-21:
I tested the feature branch, I might be missing something but how do I read
in the data? In the first notebook you sent me there was a function sweep
that returned the field data. Additionally, how can I access the nyquist?

kmuehlbauer, 2018-11-21:
@aschueth Just sent you an updated notebook. I had to make minor changes to be compatible with the structure of the other classes.

kmuehlbauer, 2019-05-16:
@aschueth #303 is pulled in. Expect the changes to be available from wradlib version 1.4.0.

closed 2019-05-16 by kmuehlbauer

mustafaalicelik, 2020-12-31:
Hi,

I have the same issue about viewing IRIS ingest files. Can you send me the prototype too?

kmuehlbauer, 2020-12-31:
@mustafaalicelik This should work from wradlib version 1.4.0. Please check with any later version and report any problems as new issue.

mustafaalicelik, 2021-01-01:
https://docs.wradlib.org/en/stable/notebooks/fileio/wradlib_radar_formats.html
here is an example for Vaisala Sigmet RAW but there is no plotting for that example.
How can I continue with that example and plot my RAW file?