wradlib/wradlib #308: import error with newer GDAL versions (with solution/workaround)

IreneCrisologo, 2019-01-06:
Not sure if this fits here in the wradlib issues because it seems more of a GDAL problem but I'm not sure where to put it, and others might encounter the same problem.

I installed wradlib in a fresh environment, but couldn't import it in jupyter notebook. The following error occurs:

```
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-3-158d7e069e94> in <module>
----> 1 import wradlib

~\Anaconda3\envs\wraderror\lib\site-packages\wradlib\__init__.py in <module>
     30 
     31     # import subpackages
---> 32     from . import adjust  # noqa
     33     from . import atten  # noqa
     34     from . import classify  # noqa

~\Anaconda3\envs\wraderror\lib\site-packages\wradlib\adjust.py in <module>
    116 
    117 # wradlib modules
--> 118 import wradlib.ipol as ipol
    119 import wradlib.util as util
    120 

~\Anaconda3\envs\wraderror\lib\site-packages\wradlib\ipol.py in <module>
     43 import warnings
     44 
---> 45 from . import util as util
     46 
     47 

~\Anaconda3\envs\wraderror\lib\site-packages\wradlib\util.py in <module>
     29 import numpy as np
     30 from scipy.ndimage import filters
---> 31 from osgeo import gdal, ogr
     32 from scipy.signal import medfilt
     33 

~\Anaconda3\envs\wraderror\lib\site-packages\osgeo\__init__.py in <module>
     19                 fp.close()
     20             return _mod
---> 21     _gdal = swig_import_helper()
     22     del swig_import_helper
     23 else:

~\Anaconda3\envs\wraderror\lib\site-packages\osgeo\__init__.py in swig_import_helper()
     15         if fp is not None:
     16             try:
---> 17                 _mod = imp.load_module('_gdal', fp, pathname, description)
     18             finally:
     19                 fp.close()

~\Anaconda3\envs\wraderror\lib\imp.py in load_module(name, file, filename, details)
    240                 return load_dynamic(name, filename, opened_file)
    241         else:
--> 242             return load_dynamic(name, filename, file)
    243     elif type_ == PKG_DIRECTORY:
    244         return load_package(name, filename)

~\Anaconda3\envs\wraderror\lib\imp.py in load_dynamic(name, path, file)
    340         spec = importlib.machinery.ModuleSpec(
    341             name=name, loader=loader, origin=path)
--> 342         return _load(spec)
    343 
    344 else:

ImportError: DLL load failed: The specified module could not be found.
```

Comparing it to an existing working environment installed in the same way (on a different machine), I noticed that the GDAL versions were different.

So I made a new environment again and this time forced GDAL to version 2.2.4, which was the version in the working environment. It solved the issue.

kmuehlbauer, 2019-01-06:
@IreneCrisologo How did you install this environment?  Did you install `wradlib` and `jupyter` in the same command? I'm trying to reproduce.

kmuehlbauer, 2019-01-06:
Under `linux` I can't reproduce this with:

```bash
conda create -n wradlib37 python=3.7 wradlib jupyter
```
or

```bash
conda create -n wradliberr36 python=3.6 wradlib jupyter
```

There are sometimes problems when upgrading/ installing into an existing environment, but I haven't seen this with a fresh environment.

IreneCrisologo, 2019-01-06:
First I created an environment:

`conda create --name newenv python=3.7`

Then activated it. Inside that environment, I installed several modules in the same command, like so:

`conda install -c conda-forge wradlib pandas pytables jupyter hdf5=1.10.2`

I also had to force an hdf5 version because I had an issue with a recent one and this version worked with an existing environment in another PC. 

The issue didn't come up when I added `gdal=2.2.4` to the install command above in another new environment.

kmuehlbauer, 2019-01-06:
@IreneCrisologo What issues do you have with latest hdf5?  I'll suspect some windows specific issues here. Need to test on windows.

IreneCrisologo, 2019-01-06:
I suspect it is a windows issue, too. I'm having these issues on a Win10 machine. In fact there is another issue with jupyter not mapping the kernel to the correct environment, such that I have to manually edit the kernel.json file to point to the python.exe in the desired environment path.

I should probably just switch my laptop to Linux...

kmuehlbauer, 2019-02-22:
@IreneCrisologo I'm going to close this. Please reopen, if this needs further handling.

closed 2019-02-22 by kmuehlbauer