wradlib/wradlib #309: Help in understanding units in FX product

guidocioni, 2019-01-16:
I'm trying to read and plot radar forecast from the DWD (RADVOR). I'm using the FX product (https://opendata.dwd.de/weather/radar/composit/fx/), which gives 2-hours forecast of reflectivity in intervals of 5 minutes, since the RV product, which should be the same but for rain rate, is still not available. My idea was then to convert everything into rain rate by firs applying `wrl.trafo.idecibel()` and then `wrl.zr.z_to_r()`.However the problem is the data initially read from the file appear to be way off as reflectivity peak at 150 dbZ, which is unlikely in this period of mostly stratiform rain. 

[image]

Is there something weird going on with the conversion? Of course retrieved rain rates are offscale because of that...

Here is a printout of the attributes

```python
rxdata, rxattrs = wrl.io.read_radolan_composite(fname)

{'cluttermask': array([], dtype=int64),
 'datasize': 1620000,
 'datetime': datetime.datetime(2019, 1, 15, 17, 5),
 'intervalseconds': 300,
 'maxrange': '150 km',
 'moduleflag': 2,
 'ncol': 900,
 'nodataflag': -9999,
 'nrow': 900,
 'precision': 0.1,
 'predictiontime': 120,
 'producttype': 'FX',
 'radarid': '10000',
 'radarlocations': ['asb',
  'boo',
  'ros',
  'hnr',
  'umd',
  'pro',
  'ess',
  'fld',
  'drs',
  'neu',
  'nhb',
  'oft',
  'eis',
  'tur',
  'isn',
  'fbg',
  'mem'],
 'radolanversion': '2.12.0',
 'secondary': array([], dtype=int64)}
```

kmuehlbauer, 2019-01-16:
@guidocioni 

According to the [FX docs](https://www.dwd.de/DE/leistungen/radarprodukte/formatbeschreibung_fxdaten.pdf?__blob=publicationFile&v=2) (sorry, in german), you need to convert the values (RVP6) to Reflectivity first (p. 7).

dBZ = RVP6/2 - 32.5

I'm pretty sure the following gives what you want:
```python
data = np.ma.masked_equal(rxdata, rxattrs['nodataflag'])
dbz = mdata/2.-32.5
pl.imshow(dbz)
```

guidocioni, 2019-01-16:
Hello Kai, I went through the document yesterday but I thought that `wradlib` already made this conversion automatically basing on the header of the `RADOLAN` file header, that's why I asked :) 

Unfortunately the values are too low if you use that formula.

```python
dbz = rxdata/2.-32.5
data = wrl.zr.z_to_r(wrl.trafo.idecibel(dbz), a=256, b=1.42)
data=[0.00010359, 0.03852534, 0.04530739, 0.04494154,
                   0.0236853 , 0.00550397, 0.08322495, 0.00010359,
                   0.00010359, 0.00010359, 0.01981591, 0.01279008,
                   0.00010359, 0.00010359, 0.00010359, 0.00010359,
                   0.00010359, 0.00010359, 0.00010359, 0.00010359,
                   0.00010359, 0.00010359, 0.00010359, 0.00010359,
                   0.08322495]
```

I should get something 10 times larger (order of 0.2 mm/h).

kmuehlbauer, 2019-01-16:
No, this special transformation is not done within the radolan-reader. 

With this example from the day before yesterday:
```python
fxname = '/home/kai/daten/daten/FX1901141205/FX1901141205_000_MF002'
rxdata, rxattrs = wrl.io.read_radolan_composite(fxname)
data = np.ma.masked_equal(rxdata, rxattrs['nodataflag'])
dbz = data/2.-32.5
rr = wrl.zr.z_to_r(wrl.trafo.idecibel(dbz), a=256, b=1.42)
fig = pl.figure(figsize=(12,8))
ax1 = fig.add_subplot(121)
cf = ax1.imshow(dbz)#, vmin=0.1, vmax=0.5)
cb = pl.colorbar(cf, shrink=0.6)
ax2 = fig.add_subplot(122)
cf = ax2.imshow(rr, vmax=1.0)
cb = pl.colorbar(cf, shrink=0.6)
```
I get the following image. Seems reasonable?

[image]

guidocioni, 2019-01-16:
I guess so... The problem is that it is quite hard to find some page to compare the results as most of the websites have just subjective color scales (light, moderate, heavy), without reporting the actual values. Looking here (https://www.weatheronline.co.uk/cgi-app/radar?CONT=euro&CREG=dwddg) it looks like the values should be correct. I'm now testing the algorithm again and let you know as soon as some intense rain develops.

kmuehlbauer, 2019-01-16:
@guidocioni Yes, comparing is quite hard for that.

You could compare the RX product with the connected FX at zeroth timestamp. They should yield more or less the same output.

kmuehlbauer, 2019-02-22:
@guidocioni Is your question answered, so we can close the issue?

guidocioni, 2019-02-22:
Sorry, forgot to report. Yeah, I think you can close the issue. Units seem to be fine.

closed 2019-02-22 by kmuehlbauer