wradlib/wradlib #324: CfRadial2 conventions

egouden, 2019-04-08:
Althought not explicitely mentioned in the documentation, the wradlib xarray reader is not agnostic and based on the "WMO" CfRadial2 conventions, which extend the NetCDF Climate and Forecast (CF) Metadata Conventions.
I think having a single convention is very useful for the community.  
However it is unclear to me why there is this distinction between metadata "variables" and "attributes".
Currently in the xarray implementation the "metadata" variables are mixed with the actual variables.
This is a bit contracdictory to the xarray and netcdf philosophy of having the actual data plus additional metadata.
I would prefer having the CF "metadata" variables in xarray attributes.
What do you think?

kmuehlbauer, 2019-04-08:
@egouden Thanks for raising that question.

Can you explain which metadata should be an attribute?

As the internal representation is based on xarray, we have to handle the pros and cons of xarray. One thing is, that xarray can only handle one subgroup per open_dataset call. It also distinguishes between Variables and Coordinates (which are variables too).

I'm absolutely in favour of only one standard, and the implementation is supposed to read and write from the standard. If it doesn't apply to the standard when reading/writing, we will naturally fix this.

kmuehlbauer, 2019-04-09:
@egouden After thinking a bit about the issue I *think* I understand the bottomline.

Please correct me if I'm wrong, you want to have **metadata** variables, like time(time), range(range), elevation(time), azimuth(time), sweep_number, sweep_mode, follow_mode, prt_mode, sweep_fixed_angle separated from the **variables** by using xarray `attrs`?

This might be possible, but only for those metadata variables which are not used as coordinates (those without a dimension connected to it). Variables like time(time), range(range) etc, have to be variables declared as `coordinates` to be used in the xarray machinery (indexing and selection etc.).

We would also have to take into consideration the behaviour of xarray machinery (eg. dropping of attributes) for several operations.

egouden, 2019-04-09:
This is not an easy discussion and it would be nice to hear from other people including CfRadial2 developers ;-)
We should probably keep the current implementation because it is consistent with CfRadial2 conventions.
Note that these conventions are under a WMO framework and are apparently not yet validated.

However, as said above, I don't get the usage of "netcdf variables" for most metadata in these conventions (is it because netcdf key-value attributes are too limited?)
I fear that having many variables with xarray could lead to inneficiencies in the data processing.
I would prefer to use only the field and coordinate variables indeed.

Back to my first remark.
Why are the CfRadial2 conventions enforced instead of the ODIM conventions.
The ODIM metadata could be put in xarray attributes and the variables would have been limited to the moments and the coordinates.
This makes it difficult for users familiar with the ODIM conventions.
Hopefully the next version of ODIM 3.0 , which is planned for 2021, will be based on CfRadial2. 
So I personally like the idea of using CfRadial2.

kmuehlbauer, 2022-11-07:
This will be tackled in the new package [xradar](https://github.com/openradar/xradar) which wradlib will depend on in the future. Please open new discussion topics over there.

closed 2022-11-07 by kmuehlbauer