wradlib/wradlib #326: xarray reader performance

egouden, 2019-04-11:
Climate studies imply the processing of decades of radar data.
It is therefore important to read the data quickly.
After diging into the code and looking at the profiling results (see below), I suggest to add the following features to the xarray interface:
- option to get a subset of the data (imagine you want to focus on a city)
- option to not decode the data
- option to not get coordinate information
- option to not get the xyz coordinates
- option to not read any extra metadata
- option to read only recommended metadata
I guess this could be easily implemented by dividing the big assign functions into smaller ones.
Attached is a pull request but just to highlight the issues(there is sitll a call to decode_cf I cannot find).

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
      2/1    0.000    0.000    0.808    0.808 {built-in method builtins.exec}
        1    0.000    0.000    0.808    0.808 <string>:1(<module>)
        1    0.000    0.000    0.806    0.806 test_io.py:1048(test_read_odim)
        3    0.018    0.006    0.802    0.267 xarray.py:1191(__init__)
        2    0.024    0.012    0.781    0.390 xarray.py:1227(assign_data_odim)
      126    0.002    0.000    0.260    0.002 conventions.py:412(decode_cf)
      126    0.003    0.000    0.156    0.001 conventions.py:351(decode_cf_variables)
      207    0.004    0.000    0.152    0.001 merge.py:392(merge_core)
      380    0.007    0.000    0.150    0.000 conventions.py:245(decode_cf_variable)
      171    0.001    0.000    0.132    0.001 common.py:295(assign_coords)
     9343    0.018    0.000    0.127    0.000 variable.py:239(__init__)
      457    0.007    0.000    0.126    0.000 common.py:171(__getattr__)
       19    0.011    0.001    0.118    0.006 polar.py:30(spherical_to_xyz)
      627    0.008    0.000    0.109    0.000 merge.py:172(expand_variable_dicts)
     2860    0.013    0.000    0.105    0.000 variable.py:41(as_variable)
      171    0.001    0.000    0.100    0.001 coordinates.py:89(update)
       59    0.000    0.000    0.091    0.002 dataset.py:3001(assign)
       78    0.000    0.000    0.089    0.001 dataset.py:2507(update)
       78    0.002    0.000    0.088    0.001 merge.py:559(dataset_update_method)
      743    0.002    0.000    0.085    0.000 dataset.py:960(__getitem__)

kmuehlbauer, 2019-04-11:
@egouden Yes, this has to be addressed for sure. Let's discuss how we can implement this in a future proof way(as good as we can imagine).

I'll have a look at the PR ASAP.

kmuehlbauer, 2019-05-10:
@egouden I see many advantages if proper implemented. Your concerns regarding speed are true and we should really try to get the best possible performance.

According to your list: 

- **option to get a subset of the data (imagine you want to focus on a city)**
For this we would need coordinates in the xarray object, or the user need to know how to claim the correct data. But AFAIU xarray netCDF implementation uses lazy-loading (http://xarray.pydata.org/en/stable/io.html#netcdf). 

_"Data is always loaded lazily from netCDF files. You can manipulate, slice and subset Dataset and DataArray objects, and no array values are loaded into memory until you try to perform some sort of actual computation."_

- **option to not decode the data**
With `decode` you're probably referring to `decode_cf`, `decode_times` and `mask_and_scale`.
It should be no problem, to declare this as keyword arguments of the `OdimH5` __init__-function.

- option to not get coordinate information (including x,y,z)
The current implementation adds x, y, z, gr, rays, bins, azimuth and elevation coordinates to the `sweep_X` datasets. There will be quite some performance leap, if this will be done only once or not at all. This can be implemented as keyword argument too.

- option to not read any extra metadata
What is extra metadata? All metadata? So that only datasets are claimed?

- option to read only recommended metadata
This would be the metadata according CfRadial standard.

Do you have any further comments or ideas?

egouden, 2019-05-14:
Sounds good!
Here are my comments:

- Indeed lazy loading is the goal but as soon as decode functions are used the whole array is read.
The decoding should happen after lazy loading.

- OK

- OK

- Yes only datatasets

- Yes only CfRadial

I am a bit lost with the complex structure of the xarray module.
I can't say how to simplify it though ;-)
Therefore I am not able to help you with this sorry.

kmuehlbauer, 2019-05-14:
@egouden AFAIU the lazy loading mechanism of xarray will not touch data arrays (even if [`mask_and_scale` is applied](http://xarray.pydata.org/en/stable/generated/xarray.decode_cf.html#xarray.decode_cf)) unless the data is claimed for processing. To be really sure I have asked on xarray-gitter channel.

First, I'd like to propagate xarray's parameters from our __init__ to `xr.decode_cf`:

- `mask_and_scale` 
  Lazily scale (using scale_factor and add_offset) and mask (using _FillValue). This would mean to have attributes (`scale_factor`, `add_offset`, `_FillValue`) connected to the `dataarrays`.
- `decode_times`
  Decode cf times (e.g., integers since ‘hours since 2000-01-01’) to np.datetime64. This would mean to have relevant attributes at the variable.
- `decode_coords`
  Use the ‘coordinates’ attribute on variable (or the dataset itself) to identify coordinates. This would mean to have the coordinate variables (`elevation azimuth range`) available.

Second, I'd like to add `georef` and `standard` keywords 

- `georef` - True/False
  Activating this switch will add 2D variables/coordinates `x`, `y`, `z`, `gr` and `rays`, and `bins` for easy georeferencing and plotting
- `standard` - 'none', 'odim', 'cf-mandatory', 'cf-full'
  - `none` - no standard-related things are extracted (plain datasets)
  - `odim` - odim-related metadata is extracted (`odim` groups) 
  - `cf-mandatory` - Mandatory cfradial metadata is extracted
  - `cf-full` - All available cfradial-standard metadata is extracted

The xarray parameters take precedence over the second keywords. If you activate `mask_and_scale` then the needed attributes will be added to the variables and mask and scale will take place even if `standard` is `none`. Same for `decode_times` and `decode_coords`.

As you mentioned, it is quite complex, because some tradeoffs in xarray package have to be overcome by added code complexity. But nevertheless the resulting improvements in data handling should pay back sufficiently.

kmuehlbauer, 2019-05-15:
@egouden If did some overhaul to the whole OdimH5-Reader in #333, keeping backwards compatibility.

The docstring of the `assign_data` function:

```
"""Assign from hdf5 data structure.

Parameters
----------
strict : bool
    If False, exports all groups verbatim into dedicated 'odim'-group.

Keyword Arguments
-----------------
decode_times : bool
    If True, decode cf times to np.datetime64. Defaults to True.
decode_coords : bool
    If True, use the ‘coordinates’ attribute on variable
    to assign coordinates. Defaults to True.
mask_and_scale : bool
    If True, lazily scale (using scale_factor and add_offset)
    and mask (using _FillValue). Defaults to True.
georef : bool
    If True, adds 2D AEQD x,y,z-coordinates, ground_range (gr) and
    2D (rays,bins)-coordinates for easy georeferencing (eg. cartopy)
standard : str
    * `none` - data is read as verbatim as possible, no metadata
    * `odim` - data is read, odim metadata added to datasets
    * `cf-mandatory` - data is read according to cfradial2 standard
      importing mandatory metadata
    * `cf-full` - data is read according to cfradial2 standard
      importing all available cfradial2 metadata (not fully
      implemented)
"""
```
Some calling examples:
```
import wradlib as wrl
data1 = wrl.io.OdimH5(filename, standard='none', mask_and_scale=False, decode_times=False, decode_coords=False, georef=False)
data2 = wrl.io.OdimH5(filename, standard='odim`, mask_and_scale=True)
data3 = wrl.io.OdimH5(filename, standard='odim`, mask_and_scale=True, decode_times=True)
data4 = wrl.io.OdimH5(filename, standard='odim`, mask_and_scale=True, decode_times=True, decode_coords=True)
data5 = wrl.io.OdimH5(filename)
data6 = wrl.io.OdimH5(filename, mask_and_scale=False)
data7 = wrl.io.OdimH5(filename, mask_and_scale=False, decode_times=False)
data8 = wrl.io.OdimH5(filename, mask_and_scale=False, decode_times=False, decode_coords=False)
```

- data1 would be the most simple dataset without any metadata nor data tampering.
- data2 would be like 1 but with masked and scaled variables
- data3 would be like 2 but with decoded time variable/coord
- data4 would be like 3 but with assigned coords according to cfradial2 and source data
- data5 would be a `cf-mandatory` style dataset
- data6 would be like 5 but with raw variables (no masking/Scaling)
- data7 would be like 6 but with undecoded time variable/coord
- data8 would be like 7 but with unassigned coordinates according to cfradial2 and source data

If tests pass, I'll merge this into master. That way others can more easily follow up. There are quite some improvements possible through this enhancements. Nevertheless the code still needs treatment.

PR #333 (2019-05-15): ENH: parametrize xarray loading
- add keywords for parametrized reading of OdimH5 data, 
- add more OdimH5 moments

closes: #326
closed 2019-05-15 by kmuehlbauer